gumyr/build123d #623: Thicken normal incorrect

MatthiasJ1, 2024-05-09:
```py
l1 = JernArc((0,0), (-1,0), 1, 180).edge().reversed()
l2 = JernArc((0,0), (1,0), 2, -90)
s = RadiusArc((0,0), (0,-1), 1)
f = sweep((l1^0)*s, l1) + sweep((l2^0)*s, l2) # Workaround for #622
```
[image]

```py
p = thicken(f, 0.4)
```
[image]

gumyr, 2024-05-10:
There is a `normal_override` parameter of thicken to help cope with these situations:
```py
p = thicken(f, 0.4, normal_override=(0, 1, 0))
```
[image]

When `thicken` was developed controlling the normal was found to be difficult so this override was added as the best solution at the time.

MatthiasJ1, 2024-05-10:
Shouldn't it just thicken in the face normal?

closed 2024-09-19 by gumyr