gumyr/build123d #627: Request: bare show_object() without relying on external editor

gordon-quad, 2024-05-16:
Would be nice to have functionality similar to https://github.com/CadQuery/cadquery/pull/1366

To be able to call `build123d.vis.show_object()` that will pop up VTK viewport without relying on any external software.

gumyr, 2024-05-16:
What is the advantage of doing:
```py
from cadquery.vis import show
```
vs.
```py
from ocp_vscode import show
```
? Is it that `cadquery.vis` gets installed with `cadquery`?

gordon-quad, 2024-05-16:
I do not have VSCode and have no intention of using it. It would be nice to be able to use build123d straight from Vim without any extra dependencies.

`cadquery.vis` is installed with cadquery, I can get my part displayed by just running `python cadquery_part.py`:

```python
from cadquery import *
from cadquery.vis import show

show(Workplane().box(1,1,1))
```

gordon-quad, 2024-05-16:
I would gladly do it myself, however i would really appreciate some guiding. If you point me out to what sort of changes need to be done with the `show` func, can help immensely.

I tried to wrap my head around what changes has been made in https://github.com/jdegenstein/jmwright-CQ-Editor vs. original https://github.com/CadQuery/CQ-editor however formatting changes there do not help with comprehension. Also not sure what should I replace the stuff imported from .occ_impl with.

gumyr, 2024-05-19:
There is a section in the docs on inter-working with CadQuery here: https://build123d.readthedocs.io/en/latest/tips.html#build123d-cadquery-integration.  As both tools share the OCP layer (the Python wrapper of the OpenCascade - OCCT - library) the `show_object` can't tell the difference between build123d and CadQuery if presented with the `wrapped` (i.e. the OCP) object.

There are four viewers listed in the docs here: https://build123d.readthedocs.io/en/latest/external.html#editors-viewers  There are others like blendquery (https://github.com/uki-dev/blendquery) which aren't listed (I'm not sure build123d support is complete with blendquery and will be happy to add it to the docs) so there are no shortage of viewer solutions. 

If one wants to use vim (I've used vi/vim for almost four decades now) this can be done by simply using ocp_vscode as the viewer and editing the code with another tool - when `show` is executed the VSC window will be updated. I use this when doing a quick visual check of the examples by running a bash loop to execute all the python programs in the examples directory and see them flash on the VSC viewer. However, I would not encourage development with just vim, an IDE provides so much additional information when simply hovering over an object that programming is so much more efficient - but each to their own.
[image]
 

As the viewer can be a lot of work (both in development and support) I'm not in a position to take on this challenge but I'd be happy to have another viewer option for users. As an illustration, bernhard (ocp_vscode) is currently working on a native C++ implementation of the tessellation step which should result in a significant performance improvement.   

To handle build123d more natively, there are only a couple things to consider:
1) The `topology.py` and`geometry.py` modules define the basic object classes like `Solid`, `Wire`, `Face` and `Plane`.  All of these objects have a `wrapped` attribute which contains the OCP object.  Therefore, if `hasattr(shape, "wrapped") ` is True pass this `wrapped` object to the viewer.
2) Builders all have a `_obj` attribute that is mapped to `part`, `sketch` or `line` for BuildPart, BuildSketch and BuildLine Builders.  Thus the same type of `hasattr` check can be used to find the OCP object from builders.  This would allow users to display a builder instance by just doing a `show(builder_instance)`.  Sketches are actually more complex as there are both local and global views of the sketch(es) but that could be a refinement to add later.

I appreciate your trying to improve the experience of users by having a built-in viewer. Unfortunately, there is currently a quite steep learning curve for new users.

gumyr, 2024-11-07:
The ocp_vscode viewer is now able to pipe the view to a web browser thus eliminating the need for vscode entirely.

closed 2024-11-07 by gumyr