gumyr/build123d #630: ExportSVG() is broken

gordon-quad, 2024-05-18:
ezdxf: 0.16.5
python: 3.11
build123d: dev branch

```python
from build123d import *

svg = ExportSVG()
```

results in:
```
Traceback (most recent call last):
  File "/home/gordon/mysrc/cadquery/export.py", line 3, in <module>
    svg = ExportSVG()
          ^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/build123d/exporters.py", line 944, in __init__
    self.add_layer(
  File "/usr/lib/python3.11/site-packages/build123d/exporters.py", line 991, in add_layer
    layer = ExportSVG._Layer(
            ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/build123d/exporters.py", line 907, in __init__
    self.fill_color = convert_color(fill_color)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/build123d/exporters.py", line 902, in convert_color
    if isinstance(c, RGB):
       ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/typing.py", line 1328, in __instancecheck__
    return self.__subclasscheck__(type(obj))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/typing.py", line 1331, in __subclasscheck__
    raise TypeError("Subscripted generics cannot be used with"
TypeError: Subscripted generics cannot be used with class and instance checks
```

gumyr, 2024-05-19:
Try upgrading ezdxf - I think that will fix the problem.

gordon-quad, 2024-05-19:
Yep, it works with 1.3.0. Did not check dependencies from `pyproject.toml` carefully, sorry.

closed 2024-05-19 by gordon-quad

gumyr, 2024-05-19:
The better solution is to add a minimum to the `pyproject.toml` for `ezdxf` but I haven't done that yet. I think this is the first dependency that's had this type of problem.