gumyr/build123d #631: Can not install - error: externally-managed-environment

gaellafond, 2024-05-21:
I follow the instructions to install it on my Ubuntu 24.04 system, but it's giving me an error:
```
$ pip install build123d
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

I also tried with pipx
```
$ pipx install build123d
Note: Dependent package 'numpy' contains 1 apps
  - f2py
Note: Dependent package 'ezdxf' contains 1 apps
  - ezdxf
Note: Dependent package 'fonttools' contains 4 apps
  - fonttools
  - pyftmerge
  - pyftsubset
  - ttx
Note: Dependent package 'numpy-stl' contains 3 apps
  - stl
  - stl2ascii
  - stl2bin
Note: Dependent package 'ipython' contains 2 apps
  - ipython
  - ipython3
Note: Dependent package 'pygments' contains 1 apps
  - pygmentize

No apps associated with package build123d. Try again with '--include-deps' to include apps of dependent packages, which are listed
above. If you are attempting to install a library, pipx should not be used. Consider using pip or a similar tool instead.
```

What am I missing?

gaellafond, 2024-05-21:
I think I figured it out. Python now requires each software to be installed in isolation using virtual environments (venv)
```
$ sudo apt-get install python3-pip python3-venv
$ python3 -m venv build123d-venv
$ source build123d-venv/bin/activate
$ pip install build123d
```

The installation worked.

I'm not sure what to do next. I was expecting an editor. The documentation says this is supposed to be a replacement for other 3D programs such as OpenSCAD. Where is the GUI?

jdegenstein, 2024-05-21:
build123d is the CAD modeling library and the editors/viewers are separate projects. The recommended editor + viewer is VSCode with the OCP CAD Viewer extension as discussed on this page of the docs https://build123d.readthedocs.io/en/latest/external.html

As mentioned on the installation page of the docs, the OCP CAD Viewer extension has the ability to install build123d for you.

closed 2024-05-21 by jdegenstein

h3n3b0sch, 2024-10-05:
For me, installing in a virtual environment doesn't work with jmwright-CQ-Editor because it doesn't recognize the installation of build123d.  
My solution was to `sudo mv /usr/lib/python3.XX/EXTERNALLY-MANAGED /usr/lib/python3.XX/EXTERNALLY-MANAGED.old` (where XX is the versionnumber) so this file is not found anymore and then install build123d the development way (from git).  
Since I don't like vscode it is now possible to use neovim or any other editor to do your programming (not that I'm any good at it...)  
  
h3n3