gumyr/build123d #633: Add major_radius bool to HexLocations

jdegenstein, 2024-05-30:
Currently `RegularPolygon` supports bool `major_radius` that defaults to `True`. In order to place hexagonal regular polygons with sides touching with the default option, one must use math to convert the major radius to minor radius (aka the apothem). I propose adding `HexLocations(... major_radius: bool = False)` which will preserve the existing default behavior, while allowing for better interoperability between `RegularPolygon` and `HexLocations`.

```py
with BuildSketch() as s:
    with HexLocations(1,3,3):
        RegularPolygon(1,6)
```
[image]

gumyr, 2024-06-01:
If one does this:
```py
with BuildSketch() as s:
    with HexLocations(1, 3, 3):
        RegularPolygon(1, 6, major_radius=False)
```
the hexes touch each other.  Is the problem that you want to work with a major radius for both `HexLocations` and `RegularPolygon`?

jdegenstein, 2024-06-01:
Yes, that is the use-case that I would like to make a bit easier.

closed 2024-06-20 by jdegenstein