gumyr/build123d #634: deprecation warning on export_stl

beep-beep-morales, 2024-06-02:
Following the documentation, I have the following code:

```
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0
center_hole_dia = 22.0

with BuildPart() as ex2:
    Box(length, width, thickness)
    Cylinder(radius=center_hole_dia / 2, height=thickness, mode=Mode.SUBTRACT)

ex2.part.export_stl("file.stl")
```

Which results in a deprecation warning:

> The method "export_stl" in class "Shape" is deprecated
>   Use the `export_stl` function instead

But the code as written above seems to be what is recommended in the note at the beginning of "Introductory examples" section of the documentation:

> If you want to save your resulting object as an STL from builder mode, you can use e.g. ex15.part.export_stl("file.stl").

Confused as to how I'm supposed to be exporting.

nobkd, 2024-06-02:
If I read the warning correctly, you should use the function `export_stl(x, ...)` instead of `y.export_stl(...)`

Searched through the Discord channel:
[image]
source: https://discord.com/channels/964330484911972403/1074840524181217452/1226331307294851122

beep-beep-morales, 2024-06-02:
OK, I'm still lost.

The following code works with no warnings:

```
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0
center_hole_dia = 22.0

with BuildPart() as ex2:
    Box(length, width, thickness)
    Cylinder(radius=center_hole_dia / 2, height=thickness, mode=Mode.SUBTRACT)

export_step(ex2.part, "block.step")
```

The following code produces an stl file but results in a deprecation warning:

```
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0
center_hole_dia = 22.0

with BuildPart() as ex2:
    Box(length, width, thickness)
    Cylinder(radius=center_hole_dia / 2, height=thickness, mode=Mode.SUBTRACT)

ex2.part.export_stl("file.stl")
```

The following code  does not work, and results in an error 'export_stl' is not defined:

```
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0
center_hole_dia = 22.0

with BuildPart() as ex2:
    Box(length, width, thickness)
    Cylinder(radius=center_hole_dia / 2, height=thickness, mode=Mode.SUBTRACT)

export_stl(ex2.part, "black.stl" )
```

beep-beep-morales, 2024-06-02:
Sorry, I think I figured it out. Installing on Windows with the following results in the above problems:

> pip install build123d

However, installing with the following now works as expected:

> python -m pip install git+https://github.com/gumyr/build123d

The first command must have installed a version that had the deprecation warning but did not have the new function implemented?

gumyr, 2024-06-02:
Thanks for pointing this out.  I've changed the docs to use the `export_stl` function.  Unfortunately I forgot to update the `__init__.py` file when I made this change and a release was created before I realized it so this is a little messed up.  The next release (which should happen before long) should clean things up along with this docs change.

beep-beep-morales, 2024-06-07:
Thank you, looking forward to the next release!

closed 2024-06-07 by beep-beep-morales