gumyr/build123d #655: package versions have conflicting dependencies, probably with cadquery or cq-editor

MotoCyti, 2024-07-01:
Thank you for building Build123d.
FWIW, I had an issue on install, but worked around it.
But Not able to install build123 (prmopt> pip install build123d) due to package versions have conflicting dependencies.
I only have cadquery and cq-editor installed on that machine, so conflict is likely with one of them.
The packages listed by pip that have the error are all ipython 8....
There are about 30 to 40 of them.  

So I downloaded an older version of python and ran that specific pip.  (in windows powershell: > Start-Process -FilePath ". . . \Python\Python311\Scripts\pip3.11.exe").  (cadquery is installed in my python 3.12 environment)
No errors on install,  but when I test in cq-editor
    import sys
    sys.path.append('C:/Users/martinez/AppData/Local/Programs/Python/Python311/Lib/site-packages')
    from build123d import *
    length, width, thickness = 80.0, 60.0, 10.0
    ex1 = Box(length, width, thickness)
    show_object(ex1)
I get this error:
    INFO: build123d: None context requested by Box

Any thoughts on what I need to fix?

Thank you.

--Chris

gumyr, 2024-07-02:
If you're using a Mac with Apple Silicon there are special instructions here: https://build123d.readthedocs.io/en/latest/installation.html#special-notes-on-apple-silicon-installs

jdegenstein, 2024-07-06:
> INFO: build123d: None context requested by Box

That is not an error, that is an info message in the log. It looks like it might be working correctly. Can you test your installation per the instructions at this link? https://build123d.readthedocs.io/en/latest/installation.html#test-your-build123d-installation

jdegenstein, 2024-09-03:
@MotoCyti were you able to solve this issue, and if so can you please close it?

MotoCyti, 2024-09-04:
I am no longer using build123d, so I don't need it fixed.  But I have no
idea if it is still a problem or not for others.

Br,
Chris

closed 2024-09-09 by jdegenstein