gumyr/build123d #662: Adding a STEP-imported part fails with pickle error

NelisW, 2024-07-22:
Windows 11  
Python 3.12.3 in a conda environment set up for build123d   
`python                    3.12.3               h1d929f7_1`

Using the latest build123d github version as of about four weeks ago.    
`build123d                 0.5.1.dev62+g4561105          pypi_0    pypi`

```
import build123d as b3d
import ocp_vscode as ov
import numpy as np
ov.set_port(3939)
ov.set_defaults(reset_camera=ov.Camera.KEEP, ortho=True, black_edges=True,measure_tools=True)

snaps = b3d.import_step("Multiboard Tight Snap.step")
with b3d.BuildPart() as snap:
  b3d.add(snaps)
```

for the STEP file downloaded from here:  
https://thangs.com/designer/Keep%20Making/3d-model/Multiboard%20Tight%20Snap-974403

(Much!) earlier version of build123d worked for STEP files downloaded from the multiboard thangs pages.

As another test, I created a simple cube in FreeCAD and exported a STEP file using checked `Export invisible objects`, and AP214 standard, and got the same error.  So on at least two differently-sourced files the error remains.
[Unnamed-PartCube.zip](https://github.com/user-attachments/files/16328815/Unnamed-PartCube.zip)

Both resulted in this error message

```
$ python HAGreen.py                                                                                                                                  
Traceback (most recent call last):                                                                                                                   
  File "K:\WorkP\3dprint\3dprint\projects\kameelperd\homeassistantgreen\HAGreen.py", line 32, in <module>                                            
    b3d.add(snaps)                                                                                                                                   
  File "C:\Users\nwillers\AppData\Local\anaconda3\envs\b123d\Lib\site-packages\build123d\operations_generic.py", line 123, in add                    
    object_iter = [obj.moved(rotation) for obj in object_iter]                                                                                       
                   ^^^^^^^^^^^^^^^^^^^                                                                                                               
  File "C:\Users\nwillers\AppData\Local\anaconda3\envs\b123d\Lib\site-packages\build123d\topology.py", line 2448, in moved                           
    shape_copy: Shape = copy.deepcopy(self, None)                                                                                                    
                        ^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                    
  File "C:\Users\nwillers\AppData\Local\anaconda3\envs\b123d\Lib\copy.py", line 143, in deepcopy                                                     
    y = copier(memo)                                                                                                                                 
        ^^^^^^^^^^^^                                                                                                                                 
  File "C:\Users\nwillers\AppData\Local\anaconda3\envs\b123d\Lib\site-packages\build123d\topology.py", line 2315, in __deepcopy__                    
    setattr(result, key, copy.deepcopy(value, memo))                                                                                                 
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                  
  File "C:\Users\nwillers\AppData\Local\anaconda3\envs\b123d\Lib\copy.py", line 151, in deepcopy                                                     
    rv = reductor(4)                                                                                                                                 
         ^^^^^^^^^^^                                                                                                                                 
TypeError: cannot pickle 'OCP.TDF.TDF_Label' object                                                                                                  
                                                                                                                                                     
```
I know too little to add any insight to this.

thanks for working so hard on b123d!

gumyr, 2024-07-22:
This is a strange problem - thanks for pointing it out.  Not as a fix but a potential work around, try extracting a solid like this:
```py
p = import_step("Issue662-PartCube.step")
print(p.label)
print(p.color)
with BuildPart() as snap:
    add(p.solid())
```
(The step file is the `Unnamed-PartCube.step` renamed)
```
Cube
Color: (0.009721217676997185, 0.009721217676997185, 0.009721217676997185, 1.0) ~ GRAY10
```
[image]

mbugert, 2024-07-23:
Ah, just wanted to report the same issue.
Test case:
```
class ImportSTEP(unittest.TestCase):

    [...]

    def test_move_single_object(self):
        export_step(Solid.make_box(1, 1, 1), "test.step")
        box = import_step("test.step")
        box_moved = Pos(X=1) * box
        self.assertEqual(tuple(box_moved.location.position), (1, 0, 0))
```
I tried to avoid assigning the `for_construction` attribute in `import_step`, but my attempts weren't fruitful.

> thanks for working so hard on b123d!

+1!

NelisW, 2024-07-23:
Works for me, thanks for your time!

gumyr, 2024-07-25:
Fixed by @mbugert  - thanks to all involved!

closed 2024-07-25 by gumyr