gumyr/build123d #664: Failed DoubleTangentArc

gumyr, 2024-07-24:
```py
with BuildLine() as h:
    l1 = PolarLine((0, 14), 2.5, 15)
    l2 = JernArc(l1 @ 1, l1 % 1, 5, -220)
    l3 = DoubleTangentArc((0, 0), (1, 1), l2)
```
```
RuntimeError: No double tangent arcs found
```

bikemike, 2024-10-06:
There is a bug in the DoubleTangentArc function that multiplies the non-unit vector perpendicular_bisector by the radius. This produces an incorrect magnitude. The perpendicular bisector should be normalized first so that the center is computed properly:

```
        for angle in [90, -90]:
            perpendicular_bisector = arc_tangent.rotate(rotation_axis, angle).normalized()

```

Also, the max_size(35.7522525416397) needs to be larger for the minimize to find a radius that works (62.54312695312509).

Another issue I have with this DoubleTangentArc is that, unlike the other Arc functions, it returns a Compound instead of a Wire. The point_at_param function does not work with compounds. Should I open a bug for that?  And it may also be nice if there was an option to Keep.BOTH.

gumyr, 2024-10-06:
Thanks!
[image]
```py
print(l3.show_topology())
```
```
Wire           at 0x733ee175a630, Center(3.354080439382976, 2.908428453244241, 0.0)
└── Edge       at 0x733ee17eff70, Center(3.354080439382976, 2.908428453244241, 0.0)
    ├── Vertex at 0x733ee175ae30, Center(-2.1316282072803006e-14, -2.1316282072803006e-14, 0.0)
    └── Vertex at 0x733ee175acf0, Center(6.70816087876598, 5.8168569064885105, 0.0)
```

closed 2024-10-06 by gumyr