gumyr/build123d #671: Step import fails to position parts correctly.

gumyr, 2024-08-17:
The step file: https://github.com/tpaviot/pythonocc-demos/blob/master/assets/models/as1-oc-214.stp appears as:
[image]
in FreeCAD but
[image]
in build123d.

There are two `l-bracket-assembly` objects but they are in the same location.

jdegenstein, 2024-08-19:
[image]

Here is a screenshot of the topology tree in Mayo which properly imports the file

gumyr, 2024-08-21:
A big thanks to @bernhard-42 for an alternate implementation and @jdegenstein for helping to debug the issue. It should be fixed now.

closed 2024-08-21 by gumyr