gumyr/build123d #676: Wire.trim doesn't really work

MatthiasJ1, 2024-08-27:
Anything but the simplest case fails. For example
```py
w = Rectangle(1,1).wire()
w = w.trim(0.5, 1)
```

gives RuntimeError: Invalid trim result.

Changing the range to (0, 0.5) doesn't error but gives a wrong result. Trimming a wire composed of many small edges (which works everywhere else) results in floating precision errors.

Also, ideally something like wire.trim(0.5, 1.5) should return the same wire with FirstParameter/LastParameter changed for closed wires (this currently works for trimming a circle wire but I haven't been able to get any other wire to trim for further testing).

gumyr, 2024-08-29:
This problem is _mostly_ fixed now.  There is still strange behavior on Arm based Macs with this test case:
```py
        p = RegularPolygon(10, 20).wire()
        t7 = p.trim(0.1, 0.2)
        self.assertAlmostEqual(p.length * 0.1, t7.length, 5)
```
which works on all other systems.  Looks like there is precision problem with one of the points on the Wire is not on the appropriate Edge.

@MatthiasJ1 is this the "many small edges" problem you noted and are you testing with an Arm based Mac?

jdegenstein, 2024-08-30:
Arm behavior is fixed now in a recent commit yesterday.

closed 2024-08-30 by gumyr

MatthiasJ1, 2024-09-01:
Sorry for the delay, just confirmed this is working, thanks!