gumyr/build123d #688: labels seem to be ignored on faces

0xorial, 2024-09-06:
I want to mark a face with a label to be able to find it easily later on. But it seems like a new Face returned every time when I call .faces()...

```
x = loft([Plane.XY * Sketch([Rectangle(1, 1)]),
          Plane.XY.offset(2) * Sketch([Rectangle(3, 3)])])
f = x.faces()[0]
f.label = 'face_label'
print(f.label)
print(x.faces()[0].label)
```

This outputs 'face_label' once, instead of twice...

gumyr, 2024-09-06:
This is by design.  Calling `faces()` creates new build123d `Face` objects from another shape.  When `f.label` is assigned this is working with an existing build123d `Face` object which has the `label` property.  For `x.faces()[0].label` to refer to `f.label` would require updating the OCCT CAD kernel's topological objects (which don't have a `label` property) which is far beyond the scope of the build123d project.

0xorial, 2024-09-06:
That makes sense, but perhaps the documentation can be a bit more clear. Currently it says 

> In order keep track of objects one can assign a label to all [Shape](https://build123d.readthedocs.io/en/latest/direct_api_reference.html#topology.Shape) objects.

and 

> Build123d’s implementation of deepcopy() for the [Shape](https://build123d.readthedocs.io/en/latest/direct_api_reference.html#topology.Shape) class (e.g. Solid, Face, etc.)

Aside from that I also saw that constructor of Face takes label.
This creates an impression that label is applicable to Face.

gumyr, 2024-09-06:
A common use case is`top = part.faces().sort_by(Axis.Z)[-1]` where `top` is then used for things like creation of a work plane  or a Face to use in a extrude or sweep. A label could be assigned as follows `top.label = "something"` but this label can't propagate back to `part` which is why only one label is generated in your example.

closed 2024-09-07 by 0xorial