gumyr/build123d #708: param_at_point fails for CenterArc

bikemike, 2024-09-23:
```
ca = CenterArc((0,0),fruit_picker_adjustable_connector_radius,-200,220)
for i in range(0,101):
	param_expected = i/100
	param_calculated = ca.param_at_point(ca@param_expected)
	print ("Expected: " + str(param_expected)  + ", Calculated: " + str(param_calculated))
	
```
Output:

```
Expected: 0.01, Calculated: -0.01571428571428564
Expected: 0.02, Calculated: -0.031428571428571465
Expected: 0.03, Calculated: -0.047142857142857104
Expected: 0.04, Calculated: -0.06285714285714275
Expected: 0.05, Calculated: -0.07857142857142856
Expected: 0.06, Calculated: -0.09428571428571421
Expected: 0.07, Calculated: -0.11000000000000001
Expected: 0.08, Calculated: -0.12571428571428567
Expected: 0.09, Calculated: -0.14142857142857132
Expected: 0.1, Calculated: -0.1571428571428571
Expected: 0.11, Calculated: -0.17285714285714276
Expected: 0.12, Calculated: -0.18857142857142842
Expected: 0.13, Calculated: -0.20428571428571424
Expected: 0.14, Calculated: -0.2199999999999999
Expected: 0.15, Calculated: -0.2357142857142857
Expected: 0.16, Calculated: -0.25142857142857133
Expected: 0.17, Calculated: -0.26714285714285696
Expected: 0.18, Calculated: -0.28285714285714264
Expected: 0.19, Calculated: -0.29857142857142843
Expected: 0.2, Calculated: -0.31428571428571406
Expected: 0.21, Calculated: -0.3299999999999999
Expected: 0.22, Calculated: -0.34571428571428536
Expected: 0.23, Calculated: -0.3614285714285712
Expected: 0.24, Calculated: -0.377142857142857
Expected: 0.25, Calculated: -0.3928571428571426
Expected: 0.26, Calculated: -0.4085714285714283
Expected: 0.27, Calculated: -0.4242857142857141
Expected: 0.28, Calculated: -0.4399999999999998
Expected: 0.29, Calculated: -0.4557142857142854
Expected: 0.3, Calculated: -0.4714285714285712
Expected: 0.31, Calculated: -0.48714285714285704
Expected: 0.32, Calculated: -0.5028571428571424
Expected: 0.33, Calculated: -0.5185714285714284
Expected: 0.34, Calculated: -0.5342857142857138
Expected: 0.35, Calculated: -0.5499999999999999
Expected: 0.36, Calculated: -0.5657142857142854
Expected: 0.37, Calculated: -0.5814285714285709
Expected: 0.38, Calculated: -0.5971428571428562
Expected: 0.39, Calculated: -0.6128571428571425
Expected: 0.4, Calculated: -0.6285714285714283
Expected: 0.41, Calculated: -0.6442857142857138
Expected: 0.42, Calculated: -0.6599999999999996
Expected: 0.43, Calculated: -0.6757142857142855
Expected: 0.44, Calculated: -0.6914285714285713
Expected: 0.45, Calculated: -0.7071428571428567
Expected: 0.46, Calculated: -0.7228571428571425
Expected: 0.47, Calculated: -0.738571428571428
Expected: 0.48, Calculated: -0.7542857142857138
Expected: 0.49, Calculated: -0.7699999999999997
Expected: 0.5, Calculated: -0.7857142857142855
Expected: 0.51, Calculated: -0.8014285714285709
Expected: 0.52, Calculated: -0.8171428571428568
Expected: 0.53, Calculated: -0.8328571428571426
Expected: 0.54, Calculated: -0.8485714285714281
Expected: 0.55, Calculated: -0.8642857142857142
Expected: 0.56, Calculated: -0.8799999999999997
Expected: 0.57, Calculated: -0.8957142857142851
Expected: 0.58, Calculated: -0.9114285714285706
Expected: 0.59, Calculated: -0.9271428571428568
Expected: 0.6, Calculated: -0.9428571428571423
Expected: 0.61, Calculated: -0.9585714285714281
Expected: 0.62, Calculated: -0.9742857142857139
Expected: 0.63, Calculated: -0.9899999999999993
Expected: 0.64, Calculated: -1.0057142857142851
Expected: 0.65, Calculated: -1.021428571428571
Expected: 0.66, Calculated: -1.037142857142857
Expected: 0.67, Calculated: -1.0528571428571427
Expected: 0.68, Calculated: -1.068571428571428
Expected: 0.69, Calculated: -1.0842857142857136
Expected: 0.7, Calculated: -1.0999999999999996
Expected: 0.71, Calculated: -1.1157142857142852
Expected: 0.72, Calculated: -1.1314285714285706
Expected: 0.73, Calculated: -1.1471428571428568
Expected: 0.74, Calculated: -1.1628571428571426
Expected: 0.75, Calculated: -1.1785714285714282
Expected: 0.76, Calculated: -1.1942857142857135
Expected: 0.77, Calculated: -1.2099999999999995
Expected: 0.78, Calculated: -1.2257142857142853
Expected: 0.79, Calculated: -1.241428571428571
Expected: 0.8, Calculated: -1.2571428571428565
Expected: 0.81, Calculated: -1.272857142857142
Expected: 0.82, Calculated: -1.2885714285714278
Expected: 0.83, Calculated: -1.3042857142857136
Expected: 0.84, Calculated: -1.3199999999999996
Expected: 0.85, Calculated: -1.335714285714285
Expected: 0.86, Calculated: -1.3514285714285708
Expected: 0.87, Calculated: -1.367142857142856
Expected: 0.88, Calculated: -1.3828571428571423
Expected: 0.89, Calculated: -1.3985714285714281
Expected: 0.9, Calculated: -1.4142857142857137
Expected: 0.91, Calculated: 1.1414285714285712
Expected: 0.92, Calculated: 1.125714285714286
Expected: 0.93, Calculated: 1.1099999999999997
Expected: 0.94, Calculated: 1.0942857142857145
Expected: 0.95, Calculated: 1.0785714285714287
Expected: 0.96, Calculated: 1.062857142857143
Expected: 0.97, Calculated: 1.0471428571428572
Expected: 0.98, Calculated: 1.0314285714285716
Expected: 0.99, Calculated: 1.0157142857142858
Expected: 1.0, Calculated: 1.0

```

bikemike, 2024-09-23:
If I change code in the Edge.param_at_point function from:

```
        if self.is_closed:
            u_value = (param_value - param_min) / (self.param_at(1) - self.param_at(0))
        else:
            param_max = _project_point_on_curve(curve, self.position_at(1).to_pnt())
            u_value = (param_value - param_min) / (param_max - param_min)
```

to 

```
        u_value = (param_value - param_min) / (self.param_at(1) - self.param_at(0))
```

The results are more in line for the range 0 to .9 but not good for params greater than 0.9. I am not familiar with OCCT and don't fully understand what the function is trying to do so I can't be be of more help. 

Output:

```
Expected: 0.0, Calculated: 0.0
Expected: 0.01, Calculated: 0.009999999999999959
Expected: 0.02, Calculated: 0.02000000000000003
Expected: 0.03, Calculated: 0.02999999999999999
Expected: 0.04, Calculated: 0.039999999999999945
Expected: 0.05, Calculated: 0.05000000000000002
Expected: 0.06, Calculated: 0.05999999999999998
Expected: 0.07, Calculated: 0.07000000000000005
Expected: 0.08, Calculated: 0.08
Expected: 0.09, Calculated: 0.08999999999999996
Expected: 0.1, Calculated: 0.10000000000000003
Expected: 0.11, Calculated: 0.10999999999999999
Expected: 0.12, Calculated: 0.11999999999999995
Expected: 0.13, Calculated: 0.13000000000000003
Expected: 0.14, Calculated: 0.13999999999999999
Expected: 0.15, Calculated: 0.15000000000000005
Expected: 0.16, Calculated: 0.16
Expected: 0.17, Calculated: 0.16999999999999996
Expected: 0.18, Calculated: 0.1799999999999999
Expected: 0.19, Calculated: 0.19
Expected: 0.2, Calculated: 0.19999999999999996
Expected: 0.21, Calculated: 0.21000000000000002
Expected: 0.22, Calculated: 0.21999999999999986
Expected: 0.23, Calculated: 0.22999999999999995
Expected: 0.24, Calculated: 0.24
Expected: 0.25, Calculated: 0.24999999999999997
Expected: 0.26, Calculated: 0.25999999999999995
Expected: 0.27, Calculated: 0.27
Expected: 0.28, Calculated: 0.27999999999999997
Expected: 0.29, Calculated: 0.29
Expected: 0.3, Calculated: 0.3
Expected: 0.31, Calculated: 0.31000000000000005
Expected: 0.32, Calculated: 0.3199999999999999
Expected: 0.33, Calculated: 0.32999999999999996
Expected: 0.34, Calculated: 0.3399999999999998
Expected: 0.35, Calculated: 0.35000000000000014
Expected: 0.36, Calculated: 0.36
Expected: 0.37, Calculated: 0.36999999999999983
Expected: 0.38, Calculated: 0.37999999999999967
Expected: 0.39, Calculated: 0.38999999999999996
Expected: 0.4, Calculated: 0.4
Expected: 0.41, Calculated: 0.40999999999999986
Expected: 0.42, Calculated: 0.41999999999999993
Expected: 0.43, Calculated: 0.42999999999999994
Expected: 0.44, Calculated: 0.44000000000000006
Expected: 0.45, Calculated: 0.4499999999999999
Expected: 0.46, Calculated: 0.46
Expected: 0.47, Calculated: 0.46999999999999986
Expected: 0.48, Calculated: 0.4799999999999999
Expected: 0.49, Calculated: 0.49
Expected: 0.5, Calculated: 0.5000000000000001
Expected: 0.51, Calculated: 0.5099999999999999
Expected: 0.52, Calculated: 0.52
Expected: 0.53, Calculated: 0.53
Expected: 0.54, Calculated: 0.5399999999999999
Expected: 0.55, Calculated: 0.5500000000000002
Expected: 0.56, Calculated: 0.56
Expected: 0.57, Calculated: 0.5699999999999998
Expected: 0.58, Calculated: 0.5799999999999997
Expected: 0.59, Calculated: 0.59
Expected: 0.6, Calculated: 0.5999999999999999
Expected: 0.61, Calculated: 0.61
Expected: 0.62, Calculated: 0.62
Expected: 0.63, Calculated: 0.63
Expected: 0.64, Calculated: 0.6399999999999999
Expected: 0.65, Calculated: 0.65
Expected: 0.66, Calculated: 0.66
Expected: 0.67, Calculated: 0.6700000000000002
Expected: 0.68, Calculated: 0.68
Expected: 0.69, Calculated: 0.6899999999999998
Expected: 0.7, Calculated: 0.7000000000000003
Expected: 0.71, Calculated: 0.71
Expected: 0.72, Calculated: 0.7199999999999999
Expected: 0.73, Calculated: 0.7300000000000001
Expected: 0.74, Calculated: 0.74
Expected: 0.75, Calculated: 0.75
Expected: 0.76, Calculated: 0.7599999999999998
Expected: 0.77, Calculated: 0.7699999999999999
Expected: 0.78, Calculated: 0.78
Expected: 0.79, Calculated: 0.79
Expected: 0.8, Calculated: 0.7999999999999999
Expected: 0.81, Calculated: 0.8099999999999997
Expected: 0.82, Calculated: 0.8199999999999998
Expected: 0.83, Calculated: 0.83
Expected: 0.84, Calculated: 0.84
Expected: 0.85, Calculated: 0.8499999999999999
Expected: 0.86, Calculated: 0.8599999999999999
Expected: 0.87, Calculated: 0.8699999999999998
Expected: 0.88, Calculated: 0.88
Expected: 0.89, Calculated: 0.8899999999999999
Expected: 0.9, Calculated: 0.8999999999999999
Expected: 0.91, Calculated: -0.7263636363636365
Expected: 0.92, Calculated: -0.7163636363636369
Expected: 0.93, Calculated: -0.7063636363636364
Expected: 0.94, Calculated: -0.6963636363636369
Expected: 0.95, Calculated: -0.6863636363636367
Expected: 0.96, Calculated: -0.6763636363636367
Expected: 0.97, Calculated: -0.6663636363636366
Expected: 0.98, Calculated: -0.6563636363636368
Expected: 0.99, Calculated: -0.6463636363636367
Expected: 1.0, Calculated: -0.6363636363636366
```

bikemike, 2024-09-23:
Could this bug be related to or a duplicate of #570 ?

gumyr, 2024-09-23:
I think it is a duplicate of #570.  The problem is with the "parameters" of the curve (a layer of object below an Edge) not starting / ending at 0, 1.  Here is an improvement:
```py
fruit_picker_adjustable_connector_radius = 1
ca = CenterArc((0, 0), fruit_picker_adjustable_connector_radius, -200, 220)

# Extract the Geom_Curve from the Shape
ca_edge = ca.edge()
first: float = ca_edge.param_at(0)
last: float = ca_edge.param_at(1)
ca_curve = BRep_Tool.Curve_s(ca_edge.wrapped, first, last)

for i in range(0, 11):
    param_expected = i / 10
    # Step 2: Project the point onto the curve
    projector = GeomAPI_ProjectPointOnCurve(
        (ca_edge @ param_expected).to_pnt(), ca_curve
    )
    projectedPoint = projector.NearestPoint()
    parameter = projector.LowerDistanceParameter()
    param_calculated = (parameter - first) / (last - first)
    print(f"{param_expected=:0.3f}, {param_calculated=:0.3f}")
```
```
param_expected=0.000, param_calculated=0.000
param_expected=0.100, param_calculated=0.100
param_expected=0.200, param_calculated=0.200
param_expected=0.300, param_calculated=0.300
param_expected=0.400, param_calculated=0.400
param_expected=0.500, param_calculated=0.500
param_expected=0.600, param_calculated=0.600
param_expected=0.700, param_calculated=0.700
param_expected=0.800, param_calculated=0.800
param_expected=0.900, param_calculated=0.900
param_expected=1.000, param_calculated=-0.636
```
Note how the values are correct now - except for the last one.  More work needs to be done here.

P.S. Note how I've used f-strings here to simplify the print statement.

closed 2024-10-02 by gumyr