gumyr/build123d #711: Example for using a lambda inside `ShapeList.filter_by`

CarpeNecopinum, 2024-09-24:
Question that came up in the Discord server was how to filter faces by being parallel to a given axis, as there is no built-in filter for it:

```py
part.faces().filter_by(lambda f: abs(f.normal_at(None).dot(axis)) < 1e-6)
```

Jern noted that there are no examples showing this feature (lambda as filter)

jdegenstein, 2024-09-24:
Thank you! Would be good to add this to page https://build123d.readthedocs.io/en/latest/key_concepts.html#custom-sorting-and-filtering

gumyr, 2024-09-28:
```py
shape = Box(1, 1, 1)
f_parallel = shape.faces().filter_by(lambda f: abs(f.normal_at().dot(Axis.X.direction)) < 1e-6)
f_perpendicular = shape.faces().filter_by(Axis.X)
```
Might want to clarify that the `dot` method expects a Vector and not an Axis.

jwagenet, 2025-04-19:
This can be closed with the implementation of #959