gumyr/build123d #717: ExtensionLine offset alters dimensions

tshead, 2024-09-27:
The following example creates an arc, then dimensions it twice with ExtensionLine - once “above” the arc, and once “below”.

Expected behavior: both dimensions should be identical.

Actual behavior: the dimensions are different.  The dimension “outside” the arc is longer than the dimension “inside” the arc, making it seem that ExtensionLine is displaying the length of the actual dimension line, rather than the length of the target path.

```
from build123d import *

draft = Draft()

with BuildSketch() as sketch:
    with BuildLine() as line:
        RadiusArc(start_point=(0, 0, 0), end_point=(100, 0, 0), radius=120)
    trace(line_width=1)
    
    ExtensionLine(border=line.edges()[0], offset=-10, draft=draft)
    ExtensionLine(border=line.edges()[0], offset=10, draft=draft)

sketch
```

[image]

gumyr, 2024-10-04:
Thanks for pointing this out.  I didn't think to test length measurement of an arc.
[image]

closed 2024-10-04 by gumyr

tshead, 2024-10-04:
@gumyr: nice to have a quick response, thanks.  I’ve never actually needed an arc length dimension, but I stumbled on this while experimenting with the various dimension-related APIs.