gumyr/build123d #725: Make convex hulls work outside of the XY-Plane

CarpeNecopinum, 2024-10-03:
Currently, convex hull computation only uses the XY coordinates of the points. This makes it impossible to use convex hulls on any plane that is perpendicular to the XY-Plane.

The culprit is the line: https://github.com/gumyr/build123d/blob/dev/src/build123d/topology.py#L8421

Replacing this static selection of the XY coordinates with one that first computes a bounding box, selects the axis with the smallest size and then projects along that one would make it possible to make convex hulls work in arbitrary planes.

(As long as the points are co-planar, and the projection doesn't happen parallel to the shared plane, the convex hull is independent of the projection direction, so there is no need to use anything fancy like a PCA to fit an optimal projection plane, just choosing one of the main Planes that is not (close to) perpendicular to the shared plane of the points will be sufficient.)

CarpeNecopinum, 2024-10-03:
My current hacky workaround for this is:
```py
def hull_in_plane(edges: Iterable[Edge], plane: Plane):
    inv = plane.location.inverse()
    return plane.location * make_hull((inv * Compound(edges)).edges())
```
(A plane to project to must be explicitly provided, choose the one that is closest to parallel with the shared plane of your points)

gumyr, 2024-10-03:
All sketching is intended to be done on a local Plane.XY and positioned afterward to the target Plane as described here: https://build123d.readthedocs.io/en/latest/build_sketch.html#sketching-on-other-planes (this is a Builder description but the same applies to Algebra mode). As such, all sketching objects and operations are designed for Plane.XY.  Relocation happens automatically in Builder mode and manually (as you've done above) in Algebra mode.

CarpeNecopinum, 2024-10-03:
So this is not something that is intended to be added/changed at some point?

An MRE of my main motivation to get something like this working is 
```py
parts = Box(1,1,1) + Pos(0,3,3) * Box(1,1,1)
projection = Sketch([p & Plane.YZ for p in parts])
hull = make_hull(projection.edges()) # -> QhullError
#hull = hull_in_plane(projection.edges(), Plane.YZ) # works
rib = extrude(hull, 0.1, both=True)

show(parts, rib)
```
i.e. I want to draw a hull around already existing features
[image]

gumyr, 2024-10-03:
You'll need to sketch on Plane.XY and transfer the sketch to the target plane.  There have been extensive discussions on this topic in the past where the pros and cons were analyzed resulting in the decision that sketching on an arbitrary plane is not supported - at least for the foreseeable future.

closed 2024-10-03 by gumyr