gumyr/build123d #727: Black is the new white?

benlaurie, 2024-10-05:
Pure white objects display as black, at least on my Mac!

```python
from build123d import Axis, Compound, Cone, Location, MM, Plane, Vector, extrude, add
from ocp_vscode import show

THICKNESS = 2 * MM

CONE_HEIGHT = 3 * MM
CONE_FLAT = 2 * MM
CONE_BASE = 6 * MM  # radius
CONE_TOP = 12 * MM  # radius

def cone(where: Vector, direction: Vector) -> Compound:
    rot = Location(Plane(where + direction.normalized() * (CONE_HEIGHT / 2 - THICKNESS / 2), z_dir=direction))
    c = Cone(CONE_BASE, CONE_TOP, CONE_HEIGHT, rotation=rot)
    print(c.show_topology())
    f = c.faces().sort_by(Axis(where, direction))[-1]
    f = extrude(f, THICKNESS)
    c += f
    return c, f

c1, f1 = cone(Vector(1, 1, 0), Vector(1, 1, 1))
c1.color = (1, 1, 1)
f1.color = (.99, .99, .99)
show(c1, f1)
```

Apologies for the slightly elaborate example, it was code I had to hand. c1 displays black and f1 white.

gumyr, 2024-10-06:
This is actually a bug in ocp_vscode.  When `c1.color = (1,1,1)` is executed a tuple is assigned to the `color` attribute which should be of class `Color`. The viewer accepts a `tuple` and does conversion to a color (black) as shown here:
[image]

However, the correct way to assign a color in build123d is: `c1.color = Color(1, 1, 1)` which results in:
[image]

Should build123d raise a warning that the user isn't using `color` correctly if an object of type `Color` isn't assigned to `color`?

benlaurie, 2024-10-06:
If I use Color, everything renders as white...

benlaurie, 2024-10-06:
Ah - apparently that's what happens if you do `Color(tuple)` - you have to do `Color(*tuple)`.

It might be helpful to issue a warning if that's not too painful.

benlaurie, 2024-10-06:
Or just turn it into a Color?

bernhard-42, 2024-10-06:
In ocp_vscode the internal Color class imported from ocp-tessellate differentiates between int and float:
(1.0, 1.0, 1.0) is white because rgb floats work on the 0.0 to 1.0 scale.
(1,1,1) is nearly black, because rgb ints work on the 0 to 255 scale.

The build123d Color class treats them as the same.

If you just use the tuples, the internal Color class is used.

closed 2024-10-07 by gumyr