gumyr/build123d #738: Mesher crashes when run as a file upon exit

jdegenstein, 2024-10-16:
This is a non-critical issue since the crash occurs AFTER all of the script has run. I currently believe the crash occurs while resources are being freed up as the python interpreter exits. Here is a test-case e.g. `crashtest.py`:
```py
from build123d import *

with BuildPart() as board:
    Box(1, 2, 3)

exporter = Mesher() # generates one exception
exporter.add_shape(board.part) # generates a second exception
```
Run like:
```
$ python crashtest.py
Exception ignored in: <function Base.__del__ at 0x7f0458f61120>
Traceback (most recent call last):
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4327, in __del__
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4091, in Release
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4026, in checkError
AttributeError: 'NoneType' object has no attribute 'SUCCESS'
Exception ignored in: <function Base.__del__ at 0x7f0458f61120>
Traceback (most recent call last):
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4327, in __del__
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4091, in Release
  File "/home/oem/miniconda3/envs/ocp_vscode2/lib/python3.11/site-packages/py_lib3mf/Lib3MF.py", line 4026, in checkError
AttributeError: 'NoneType' object has no attribute 'SUCCESS'
```
I currently suspect that some logic can be added to `class Mesher` that will solve this problem (such as to free up resources as the script exits somehow). It also could be a problem with the lib3mf python bindings.

gautaz, 2024-11-04:
And I just realized that this was in fact a PEBKAC issue, I missed the `.part` in the `add_shape`.
My apologies for polluting the thread.

jdegenstein, 2025-11-20:
@gumyr FYI see the issue I just linked for the source of this problem (lib3mf via the use of AutomaticComponentToolkit)

gumyr, 2026-10-01:
Child process exits 0 with empty stderr

closed 2026-10-01 by gumyr