gumyr/build123d #742: Inheriting coordinate systems?

l29ah, 2024-10-17:
I often draw sketches made of polylines with parametrized coordinates, OpenSCAD-style, then extrude them, and then sketch other stuff on top of extrusions. It's really annoying that every time a new face is taken, a new origin of the coordinate system is produced, and all my prior calculations become invalid, so i have to fight the offsets of the new origin to apply them.

It would be nice if there were builders that don't redefine the coordinate system origin, or otherwise a way to preserve them at least for some axes.

gumyr, 2024-10-18:
If you provide an example of exactly the problem your are seeing I'll be able to provide better help. Here is my guess at the problem you might have:
```py
from build123d import *
from ocp_vscode import show_all

pts = [(0, 0), (1, 0), (2, 1), (0, 0)]
with BuildPart() as p1:
    with BuildSketch() as s1:
        with BuildLine() as outline:
            l1 = Polyline(*pts)
        make_face()
    extrude(amount=0.5)

with BuildSketch(p1.faces().sort_by(Axis.Z)[-1]) as s2:
    Circle(0.25)

show_all()
```
[image]
`s1` is using the default workplane - Plane.XY - so the `Polyline` is where one expects.  `s2` isn't using the default workplane but one created from the top face of `p1`.  When creating a `Plane` from a `Face` the origin is set to the center of that `Face` which means that the `Circle` is no longer at the global origin.

There are many ways to create a workplane that allows one to control the origin (and other parameters like the x direction), for example `Plane.XY.offset(0.5)` (the height could be derived from `p1` instead of being provided `Plane.XY.offset(p1.part.bounding_box().max.Z)`.

To summarize, `BuildSketch` needs a `Plane` which can be generated in many ways, including from a `Face` - see the [Plane](https://build123d.readthedocs.io/en/latest/direct_api_reference.html#geometry.Plane) class for more information.

gumyr, 2024-10-18:
That combination of things isn't a valid way to create a Plane but unfortunately there is no indication of that.

Try: `Plane(origin=(...), x_dir=(1,0,0), z_dir=p1.faces().sort_by(Axis.Z)[-1].normal_at())` where ... is the global position of where you want the origin to be.

closed 2024-10-22 by gumyr