gumyr/build123d #743: Create a section in the docs attempting to describe the design process

gumyr, 2024-10-18:
The documentation describes the tools available to design objects but it doesn't provide an description of the steps one might use to go about creating a design with build123d.  Here is a proposal for what this section might contain - feedback requested:

### Designing a Part in build123d

Designing a part with build123d involves a systematic approach that leverages the power of 2D profiles, extrusions, and revolutions. Where possible, always work in the lowest possible dimension, 1D lines before 2D sketches before 3D parts. Here is a guide to get started:

1. Examine the Part in All Three Orientations

    Start by visualizing the part from the front, top, and side views. Identify any symmetries in these orientations, as symmetries can simplify the design by reducing the number of unique features you need to model.

2. Identify Rotational Symmetries

    Look for structures that could be created through the rotation of a 2D shape. For instance, cylindrical or spherical features are often the result of revolving a profile around an axis. Identify the axis of rotation and make a note of it.

3. Select a Convenient Origin

    Choose an origin point that minimizes the need to move or transform components later in the design process. Ideally, the origin should be placed at a natural center of symmetry or a critical reference point on the part.

4. Create 2D Profiles

    Design the 2D profiles of your part in the appropriate orientation(s). These profiles are the foundation of the part's geometry and can often represent cross-sections of the part. Mirror parts of profiles across any axes of symmetry identified earlier.

5. Use Extrusion for Prismatic Features

    For solid or prismatic shapes, extrude the 2D profiles along the necessary axis. You can also combine multiple extrusions by intersecting or unionizing them to form complex shapes. Use the resulting geometry as sub-parts if needed.

6. Generate Revolved Features

    If any part of the geometry can be created by revolving a 2D profile around an axis, use the revolve operation. This is particularly useful for parts that include cylindrical, conical, or spherical features. Combine these revolved sub-parts with existing features using additive, subtractive, or intersecting operations.

7. Combine Sub-parts Intelligently

    When combining multiple sub-parts, keep in mind whether they need to be added, subtracted, or intersected. Subtracting or intersecting can create more refined details, while addition is useful for creating complex assemblies.

8. Apply Chamfers and Fillets

    Identify critical edges or vertices that need chamfering or filleting. Use build123d’s selectors to apply these operations accurately. Always visually inspect the results to ensure the correct edges have been modified.

9. Design for Assembly

    If the part is intended to connect with others, add features like joints, holes, or other attachment points. Ensure that these features are precisely located to ensure proper fitment and functionality in the final assembly.

10. Plan for Parametric Flexibility

    Wherever possible, make your design parametric, allowing dimensions and features to be easily adjusted later. This flexibility can be crucial if the design needs modifications or if variations of the part are needed.

11. Test Fit and Tolerances

    Visualize the fit of the part within its intended assembly. Consider tolerances for manufacturing, such as clearance between moving parts or shrinkage for 3D-printed parts. Adjust the design as needed to ensure real-world functionality.

CePeU, 2024-10-18:
Fillet early if possible this will avoid the necessity to find the correct edges later on.

Maybe make some "standard" hints like (and they might be dumb but things I "discovered"):

- Use polyline instead of a set of lines
- explain (again) the operator to place a startingpoint between 0 and 1 on a line
- Give a hint about filters and selecting faces,edges and vertices
- explain right hand rule
- explain orientation of a face/place and how it impacts extrudes
- explain the difference between brep and stl

I think these are the most often asked questions a beginner has asked in the discord channel

CePeU, 2024-10-18:
Oh yes one more! The difference between move, locate etc. a more or less more baisc introduction to handling the placing of objects. I am still struggling ;)

wcraigtrader, 2024-10-18:
Design for Manufacturing
- Consider the materials you're planning to use, and their strengths and weaknesses. 
- If you're 3D printing using FDM, consider trade-offs orienting your part: layer adhesion, layer strength, supports,
- If you're milling a part, do you have a 2-axis mill, 3-axis mill, or more? How will you secure your material during milling?

gumyr, 2024-10-21:
I've created a first version of [Designing a Part in build123d](https://build123d.readthedocs.io/en/latest/tutorial_design.html)  with a66b8c76f47d8b9bd16fcd6368654e49314a762e. It doesn't cover all of the points above as some of these are more specific to different manufacturing processes, but we should have a page on that too.

closed 2024-10-22 by gumyr