gumyr/build123d #747: Add a section to the docs describing all the ways to change object locations

gumyr, 2024-10-22:
```py
with Locations():
    ...
with Builder(workplane):
    ...
shape.move(location) # relative without copy
shape.moved(location) # relative with copy
shape.locate(location) # absolute without copy
shape.located(location) # absolute with copy
shape.translate((x,y,z)) # transform <- needs to change
shape.rotate(Axis,angle) # transform <- needs to change
shape.position = (x,y,z) # absolute without copy
shape.position += (x,y,z) # relative without copy
shape.orientation = (rx,ry,rz) # absolute without copy
shape.orientation += (rx,ry,rz) # relative without copy
```

jdegenstein, 2024-10-22:
Related to https://github.com/gumyr/build123d/issues/450

gumyr, 2025-01-25:
Complete with 9f5b4eaa67b91e37bce1ba6082dee6255a6ce463  See: https://build123d.readthedocs.io/en/latest/moving_objects.html

closed 2025-01-25 by gumyr