gumyr/build123d #748: There are major problems with the documentation. Please update the documentation

RFARREY, 2024-10-23:
It seems there's a discrepancy between the Build123d documentation and the actual implementation in version 0.7.0.

The sweep function in this version doesn't accept the section parameter, and it seems the documentation is outdated.

jdegenstein, 2024-11-01:
@RFARREY did the above address your question / concern?

KilowattSynthesis, 2024-11-05:
Whether it answers the question, there's a few of these where methods with the same name have different arguments, different argument orderings, and different semantics. It's really too bad.

jdegenstein, 2024-11-05:
@KilowattSynthesis PRs/issues are welcome to fix any inconsistencies you may find. In this case `section` vs. `sections` is an intentional difference between the two.

I am closing this issue as we have not received any response in several weeks from the author.

closed 2024-11-05 by jdegenstein

KilowattSynthesis, 2024-11-05:
Fair enough; I think I was misunderstanding the difference. Cheers!