gumyr/build123d #750: Create a tracing tool that can extract all of the OCP calls (with parameters) from buid123d code

gumyr, 2024-10-24:
To aid with debugging of OCCT issues, a tool that could extract the OCP calls (with parameter values) either statically or dynamically from build123d code would enable rapid creation of OCCT test cases.

Potentially something like this:
```py
import sys
from build123d import Box

def trace_func(frame, event, arg):
    # We're only interested in call events
    if event != "call":
        return

    try:
        # Get the name of the module where the function is defined
        module_name = frame.f_globals.get("__name__")

        # Filter for build123d calls
        if module_name and module_name.startswith("build123d"):
            func_name = frame.f_code.co_name
            func_file = frame.f_code.co_filename
            func_line = frame.f_lineno
            print(f"Calling {func_name} in {module_name} at {func_file}:{func_line}")

            # Only print arguments that are safe to inspect
            try:
                print(f"Arguments: {frame.f_locals}")
            except AttributeError:
                print(f"Could not inspect locals for {func_name}")

    except AttributeError as e:
        print(f"AttributeError caught in {func_name}: {e}")

    return trace_func

# Set the trace
sys.settrace(trace_func)

# This will trace any build123d calls, including those calling into OCP
b = Box(1, 1, 1)

# Turn off tracing
sys.settrace(None)
```
where the filtering was improved to only include OCP calls.

jdegenstein, 2024-10-25:
This is a fantastic idea!

gumyr, 2026-10-04:
Now available as: https://github.com/build123d/ocp_recorder

closed 2026-10-04 by gumyr