gumyr/build123d #752: Dimensional check is incorrect in Compound.__sub__(), other possible latent problems

eecharlie, 2024-11-01:
```
from build123d import *
shape1 = Pos(0,0,0) * Box(30,30,30)
shape2 = Pos(30,0,0) * Box(50,5,5)
shape1 - shape2                                     # This is fine and produces a box with a hole
Compound(shape1) + Compound(shape2)                 # This is fine and produces a box on a stick
Compound([shape1,shape3]) + Compound(shape2)        # This works
sl = (ShapeList([shape1]) - ShapeList([shape2]))[0] # Executes, but just returns shape1, why?

Compound(children=[shape1]) - Compound(children=[shape2])   # This throws an error, why?
Compound(shape1) - Compound(shape2)                     # This throws an error, why?

# Error: TypeError: '<' not supported between instances of 'NoneType' and 'NoneType'
# when the dimension of self & other are checked
```

There are two problems quickly apparent in reviewing the source code for `__sub__()`:
```
        for _other in others:
            if type(_other)._dim < type(self)._dim:
                raise ValueError(
                    f"Only shapes with equal or greater dimension can be subtracted: "
                    ...
```
The error language is inconsistent with the actual conditional, in that the conditional should be `<=`, though this is not why the above example fails. The example fails because `type(Compound(Box(1,1,1)))._dim` is `None` when I think it should be `3`. 

The reason that this bug exists in `__sub__` and not in `__add__` is that add only checks for *equality* of dimension, and `None == None`.

eecharlie, 2024-11-01:
Opened #754

gumyr, 2024-11-01:
A Compound is a collection of any Shape; one can build a Compound of mixed 2D and 3D shapes so the `_dim` of Compound is explicitly set to None.  The `Part`, `Sketch` and `Curve` versions of `Compound` are intended to be used for collections of 3D, 2D and 1D Shapes respectively.

However, your other findings are worth investigation.

eecharlie, 2024-11-01:
I think it’s a bit simpler than I first thought, see the PR.

eecharlie, 2024-11-01:
So, why was I taking the approach in the above failing example in the first place? I think it is because the first and only place I recall seeing an explanation of the basic classes in the documentation is here:
https://build123d.readthedocs.io/en/latest/key_concepts.html#topology which includes `Compound` but none of `Part` `Sketch` or `Curve` classes. I think that an additional possibly confusing subtlety for me is the inclusion of `Shape` at the very end - I understand that it's the base class but I don't understand what I'd do with it. Should go at the beginning of the list of things that inherit from it? 

On the analogous page for Algebra mode, https://build123d.readthedocs.io/en/latest/key_concepts_algebra.html it *mentions* `Part` `Sketch` and `Curve` but there's no context on how one would obtain instances of them except implicitly from the examples that use function like `Box` and `Cylinder` which I don't think are documented anywhere outside of the weeds of the full API documentation.

I might suggest that the Topology section of Key Concepts (Builder Mode) be pulled out and precede both Key Concepts sections, because it's relevant to both. Further, I would suggest including in a single place a brief definition of `Part` `Sketch` `Curve` and explanation of their inheritance from `Compound`. It should be made clear that both `Compound` and `Shape` are not meant for direct use.
Additionally, a really helpful section complementing Topology and prior to Algebra/Builder mode would be a quick rundown of the helpful primitive `Part` `Sketch` and `Curve` classes, such as `Box` and `Cylinder`, if it's just a laundry list of their names under the description of e.g. `Part`.

If there's some page in the documentation I missed that contains the above content, please point me to it, I've been unsuccessful with text searches for class names.

gumyr, 2024-11-01:
`Compound` is definitely intended to be used, one of the ways is for an assembly - a collection of physically disjoint components - as described here: https://build123d.readthedocs.io/en/latest/assemblies.html  An `Assembly` class will be introduced as a subclass of `Compound` to help make this use case distinction.

`Part` can contain multiple disjoint objects as well but this isn't the primary purpose.  Often in the creation of an single `Solid` a intermediate stage may contain multiple `Solids` so this is not prohibited.

I've marked this Issue with the documentation tag, some of the recommendations you make here would likely be helpful for new users.

eecharlie, 2024-11-01:
Ah, thanks. Will go back to understanding how my use case is better approached, informed by this, and see if I can understand your suggestion in the PR on minuend.

gumyr, 2024-11-01:
`Compound.unwrap` just strips extraneous layers of `Compound` from an object.  With the `unwrap(fully=True)` option, the entire `Compound` "wrapper" is eliminated if there is a single Shape in the `Compound` (which OCCT generates frequently).  As both `self` and `others` are going to be combined and a new object generated these wrappers add no value.

gumyr, 2024-11-06:
These operators handle Compounds now.

closed 2024-11-06 by gumyr