gumyr/build123d #753: Broken pip install cadquery-ocp dependency for Apple silicon platform

eecharlie, 2024-11-01:
I'm unable to get cadquery/ocp working with `build123d`. 

- I've tried straightforward pip install as below, which hangs up on finding a version of `cadquery-ocp`:
```
ERROR: Could not find a version that satisfies the requirement cadquery-ocp>=7.7.0 (from build123d) (from versions: none)
ERROR: No matching distribution found for cadquery-ocp>=7.7.0
```
- Reviewing that package's [instructions](https://github.com/CadQuery/OCP), I've tried installing via `conda install -c conda-forge -c cadquery ocp` which is successful, but does not satisfy the pip dependencies of `build123d`

However, disabling `build123d`'s dependencies on `cadquery-ocp` and `ocpsvg` (via editing `pyproject.toml`) allow installation of build123d to proceed, as in my case I have `ocpsvg==0.2.1` already installed.

Can a workaround/accommodation for installation on Apple silicon be included in the install process, even if it's specific manual instructions in the readme, so that Apple silicon users don't have to hack their way to this?

eecharlie, 2024-11-01:
Whoops, this is well known and tracked [here](https://github.com/CadQuery/ocp-build-system/issues/11)

eecharlie, 2024-11-01:
Whoops, well known and tracked [here](https://github.com/CadQuery/ocp-build-system/issues/11).

closed 2024-11-01 by eecharlie

jdegenstein, 2024-11-01:
saw that you closed already, but FYI we also have a workaround in the docs https://build123d.readthedocs.io/en/latest/installation.html#special-notes-on-apple-silicon-installs

eecharlie, 2024-11-01:
Thanks! I had been limping along on having installed `ocp` and `cadquery` but not `cadquery-ocp` but I was still getting some inline Jupyter rendering with a horrible limited UI. After following those directions, everything is broken: most unit tests fail in my local development branch of build123d, and in VS Code Jupyter notebooks I get a vtk error message and blank canvass when it tries to render.
I think I'm going to try Docker.