gumyr/build123d #758: Sweeping a circle on a helix path not working as expected

KilowattSynthesis, 2024-11-05:
Consider this example Helix, which works exactly as you'd expect.

```python
def make_helix() -> bd.Helix:
    """Make basic helix."""
    return bd.Helix(
        radius=2,
        pitch=1,
        height=10,
    )
```

[image]

Now, consider this circle swept around the Helix:

```python
def make_helix_sweep() -> bd.Part:
    """Make basic helix sweep."""
    p = bd.Part()
    p += bd.sweep(
        path=make_helix(),
        sections=bd.Circle(radius=1),
    )
    return p
```

[image]

I would have expected the output to come out as a "spring" appears (constant major diameter along the length). Why did it come out looking so different from a spring?

gumyr, 2024-11-05:
The profile needs to be perpendicular to the path to get a "spring" - like this:
```py
path = Helix(radius=2, pitch=1, height=10)
profile = (path ^ 0) * Circle(radius=0.3)
swept = sweep(path=path, sections=profile)
```
[image]
The `(path ^ 0)` operator returns a `Location` at the beginning of the path which then orients the profile to be perpendicular to the path.

KilowattSynthesis, 2024-11-05:
Neat, thank you for the support! I've created #759 and #760, which depict documentation changes which would have made that discovery easier. 

I'll close this issue in favor of those other two.

closed 2024-11-05 by KilowattSynthesis

jdegenstein, 2024-11-05:
By the way, this is why I prefer to develop outside of functions until the part is "finished" -- it is important to be able to easily view all objects. Because the function internals are not added to the global scope, it requires more care to view them using e.g. OCP CAD Viewer or CQ-editor. CodeCAD can be quite complex because of the interplay between the code structure and sometimes-not-so-trivial 3D representation.