gumyr/build123d #761: Docs: Note that the `radius` arg of `bd.Helix` is the "minimum radius" when `cone_angle>0`

KilowattSynthesis, 2024-11-05:
When `cone_angle>0`, the cone of the helix expands **outwards**. The docs don't currently mention this.

gumyr, 2024-11-09:
If the `cone_angle<0` the helix gets smaller as it proceeds so the `radius` isn't the "minimum" but maybe the "base" radius.

KilowattSynthesis, 2024-11-11:
Hmm good note; hadn't considered that. I don't think "base" radius is necessarily clear, as it's not clear what "base" means (I guess it's the radiux closest to the orgin?).

Perhaps:

> When `cone_angle != 0`, the `radius` argument is the radius at the side of the cone closest to the origin. When `cone_angle > 0`, the radius expands outwards farther from the origin; when `cone_angle < 0`, the radius spirals inwards.

commit 1b63aa34 (2025-03-27): Helix: clarify how cone_angle changes the radius to resolve #761
closed 2025-03-27 by gumyr