gumyr/build123d #763: `Wire_obj ^ 2` should throw an error, I think

KilowattSynthesis, 2024-11-05:
My understanding is that the caret operator, which is an overload for `Wire.location_at()` (or any other object which inherits from Mixin1D), accepts an argument where `0 <= distance <= 1` when `position_mode = PositionMode.PARAMETER` (the default). 

If this is true, calling the caret operator with a 2nd-operand greater than `1` should be invalid, and should raise an error instead of acting silently. 

This will help people's understanding of what the function does, and may help catch bugs as well.

gumyr, 2024-11-05:
No.  Although normally `u`, `v` values have the range of `0.0 - 1.0` one can legitimately use values outside of this range as the underlying curve object is valid beyond these end points.

KilowattSynthesis, 2024-11-05:
Maybe this is a documentation request then, which explains how that works.

What happens when one does `polyline ^ 2`? Does it just extend the closest-to-the-end line segment?

gumyr, 2024-11-05:
Let's try and find out:
```py
from build123d import *
from ocp_vscode import show_all

path1 = Polyline((0, 1, 0), (4, 1, 0), (4, 4, 0))
p1 = path1 ^ -1
p2 = path1 ^ 2

path2 = Spline((-2, 6), (0, 2), (2, 6))
p3 = path2 @ -1
p4 = path2 @ 2

show_all()
```
[image]

Within the kernel all lines are represented by an OCCT `Geom_Curve` which is "trimmed" to a given start and end point.  When a user asks for `u` values outside of `0.0-1.0` they are simply requesting  values on the base `Geom_Curve`.

Maybe a user would like to align text to a line and it goes a little off one or both ends - allowing values outside of the 0.0 to 1.0 range enables this. I don't recommend users do this unless they understand what they are doing though.

KilowattSynthesis, 2024-11-06:
Thanks, I think that's a helpful example. Would be awesome to be able to create a similar clear example that could go in the docs!