gumyr/build123d #772: Model seems misplaced regarding the origin when 3MF export is displayed with f3d

gautaz, 2024-11-10:
Hello,

Newbie here, this might be a PEBKAC or it might be related to #301.

When viewing with [f3d](https://github.com/f3d-app/f3d) what is exported in 3MF from [introductory example 15](https://build123d.readthedocs.io/en/latest/introductory_examples.html#mirroring-symmetric-geometry), I obtain this:

[image]
I forgot to display the coordinate system in the screenshot but:
* the X axis is the red one with positive values on the upper side
* the Y axis is the green one with positive values on the left side

Here is the code I am using:

```python
from build123d import *

a, b, c = 80, 40, 20

with BuildPart() as ex15:
    with BuildSketch() as ex15_sk:
        with BuildLine() as ex15_ln:
            l1 = Line((0, 0), (a, 0))
            l2 = Line(l1 @ 1, l1 @ 1 + (0, b))
            l3 = Line(l2 @ 1, l2 @ 1 + (-c, 0))
            l4 = Line(l3 @ 1, l3 @ 1 + (0, -c))
            l5 = Line(l4 @ 1, (0, (l4 @ 1).Y))
            mirror(ex15_ln.line, about=Plane.YZ)
        make_face()
    extrude(amount=c)

exporter = Mesher()
exporter.add_shape(ex15.part)
exporter.write("test.3mf")
```

The first point of `l1` should be located at `(0, 0)` but seems in fact located at `(0, -c)` in f3d.

gumyr, 2024-11-10:
It looks like some centering is going on here, but who is doing it: build123d or f3d?  Do you have some other models that you could import into f3d to see if f3d is doing this?

jdegenstein, 2024-11-10:
I did some quick tests and it appears that f3d does not respect the original coordinates of an input file. I do not recommend using f3d as a tool for developing models with build123d. Recommended viewers are on this page:  https://build123d.readthedocs.io/en/latest/external.html

gautaz, 2024-11-11:
Hello, thanks both for the input.

I did look at the recommended viewers list before but:
* ocp-vscode & PartCAD VS Code extension: sorry, old vim guy here :slightly_smiling_face: (I am not a big fan of vscode, it takes too much memory for an editor/IDE)
* cq-editor fork & yet-another-cad-viewer: not available on NixOS so I opted for an already available solution at first (hence f3d)

Anyway, as @jdegenstein said, f3d is the culprit so this issue can be closed.
I will probably open an issue on f3d if there is no way of keeping the original coordinates.

I was interested in yet-another-cad-viewer before looking at f3d and I took the easiest way.
If f3d is finally not a solution, I will perhaps take some additional time to build a container for it as I did for build123d (see [this repo](https://github.com/gautaz/cadcode)).

closed 2024-11-11 by gautaz

gautaz, 2024-11-11:
So it turns out there is an option for f3d to keep the origin where it should be: `--grid-absolute`.
Combined with the `--watch` option and a few others that can be all kept in a configuration file, f3d still seems a decent solution for previewing a model.

gumyr, 2024-11-11:
>     * ocp-vscode & PartCAD VS Code extension: sorry, old vim guy here 🙂 (I am not a big fan of vscode, it takes too much memory for an editor/IDE)

I've used vim for many decades so I know where you're coming from.  Whatever you use, having an IDE to aid you makes a big difference; here are just a couple:
- as I hover over an object the following pops up:
[image]
  which describes the complete API for this object.  This will save you countless hours as you won't need to bring up the documentation for every API.
- the ability to "Rename Symbol" has improved my code significantly.  Naming things can be quite difficult so being able to go back and change the name of something without having to worry about having a good regex to avoid mangling other symbols is super useful.

There are various vim editors that have some or all of these capabilities so you don't need to drop vim.  Using an IDE will make your programming life better.

gautaz, 2024-11-11:
I am over fifty, more than 30 years of vim, really hard to change now.
I am in fact using neovim now, I will probably die with it :rofl:.