gumyr/build123d #775: Aliases: Create alignment aliases for common anchor-based alignments

KilowattSynthesis, 2024-11-11:
I come from using OpenSCAD's BOSL2 library, and it has a few ease-of-use things that are very close, but are missing in build123d. 

Would you accept a PR that adds the following aliases? They're basically a copy of what's in here: https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#subsection-anchor

```python
"""Shortcut aliases for the Align class.

The names specify the "anchor" point. For example, "ANCHOR_BOTTOM" means to align the
bottom of the object to the origin.
"""

import build123d as bd

ANCHOR_TOP = (bd.Align.CENTER, bd.Align.CENTER, bd.Align.MAX)
ANCHOR_BOTTOM = (bd.Align.CENTER, bd.Align.CENTER, bd.Align.MIN)
ANCHOR_FRONT = (bd.Align.CENTER, bd.Align.MIN, bd.Align.CENTER)
ANCHOR_BACK = (bd.Align.CENTER, bd.Align.MAX, bd.Align.CENTER)
ANCHOR_LEFT = (bd.Align.MIN, bd.Align.CENTER, bd.Align.CENTER)
ANCHOR_RIGHT = (bd.Align.MAX, bd.Align.CENTER, bd.Align.CENTER)

ANCHOR_CENTER = (bd.Align.CENTER, bd.Align.CENTER, bd.Align.CENTER)

ANCHOR_TOP_LEFT = (bd.Align.MIN, bd.Align.MIN, bd.Align.MAX)
ANCHOR_TOP_RIGHT = (bd.Align.MAX, bd.Align.MIN, bd.Align.MAX)
ANCHOR_BOTTOM_LEFT = (bd.Align.MIN, bd.Align.MAX, bd.Align.MAX)
ANCHOR_BOTTOM_RIGHT = (bd.Align.MAX, bd.Align.MAX, bd.Align.MAX)

ANCHOR_FRONT_LEFT = (bd.Align.MIN, bd.Align.MIN, bd.Align.CENTER)
ANCHOR_FRONT_RIGHT = (bd.Align.MAX, bd.Align.MIN, bd.Align.CENTER)
ANCHOR_BACK_LEFT = (bd.Align.MIN, bd.Align.MAX, bd.Align.CENTER)
ANCHOR_BACK_RIGHT = (bd.Align.MAX, bd.Align.MAX, bd.Align.CENTER)
````

closed 2024-11-11 by gumyr

KilowattSynthesis, 2024-11-11:
Why not have both available then? Don't let perfect be the enemy of good/better.

If there's no chance you'd accept a PR like this, I'm going to have to go and create my own add-on library. I hate when ecosystems get sharded like this, as it makes publishing bug reports/tutorial content so much harder as you now need that other version of the library, and often have to sync versions between the libraries.

Are you sure that's the decision you're taking a hard stance on? That `(bd.Align.CENTER, bd.Align.CENTER, bd.Align.MAX)` is the only viable standard option?

gumyr, 2024-11-11:
I could ask you the same question.  Are you paying for each character in your file?  The use of these abbreviations is only because the user is wants the code to look different not function differently and I always go for clarity over brevity (with the exception of those speed modellers out there that use abbreviations to win).  Start using `black` for formatting and stop worrying what your code looks like. 

With respect to the terms: what does "top" and "bottom" even mean if the user is working on some non default `Plane.XY`? `Plane.YX` reverses the normal of `Plane.XY`, what is "up" or "down", "right" or "left"?  It all sounds good until one looks into the details.