gumyr/build123d #776: Aliases: Create XCyl, YCyl, and ZCyl alias methods

KilowattSynthesis, 2024-11-11:
Would you accept a PR which creates aliases for each of the three cylinder orientations by name? 

It would align with the BOSL2 library style, which is a tad bit easier to work with currently: https://github.com/BelfrySCAD/BOSL2/wiki/shapes3d.scad#module-xcyl

The big advantage of this method (instead of using `bd.Cylinder(rotation=...)` currently) is that the `align` argument will work "properly". For example, in an X-Cylinder, aligning with `align=(bd.Align.CENTER, bd.Align.CENTER, bd.Align.MIN)` would place the bottom of the rounded part at the origin, without the designer having to think about rotating the alignment coordinate system (as currently, the rotation is applied after the alignment operation).

KilowattSynthesis, 2024-11-11:
That'd be a crazy breaking change, but I'd ultimately support it. 

I fully understand that cylinders can be oriented in more than 3 ways. You can still rotate it _after_ you've started with the closest base cylinder. Imagine a cylinder pointing in the positive X direction, rotated 10 degrees about the Z axis, and aligned such that the bottom of its curve touches the XY plane. Are you telling me that you'd get that series of rotations and transform right the first time?

Compare that with a simple `bd.XCyl(radius, height, alight=bd.ANCHOR_BOTTOM).rotate(bd.Axis.Z, 10)`.

You have the opportunity to learn from/build on what makes those other tools so awesome; it's too bad that you want to force these sorts of trivial implementations into other libraries.

jdegenstein, 2024-11-11:
build123d is not trying to emulate OpenSCAD/extensions -- it is a fundamentally different software. I fear that many of your recommendations are based on a limited understanding of what methodologies are available in build123d that differ from some of the other tools you are familiar with, for instance one analogue to "YCyl":
```py
with BuildPart() as p:
    with BuildSketch(Plane.XZ) as s:
        Circle(1)
    extrude(amount=1)
```
Or alternatively using 3D primitives:
```py
with BuildPart() as p:
    with Locations(Plane.XZ):
        Cylinder(1,1)
```
And now a far less trivial example that is not possible in *SCAD/derivatives:
```py
with BuildPart() as p:
    with BuildSketch() as s:
        RegularPolygon(1,3)
    loft(sections=[s.sketch,Vertex(0,0,1)]) # create a triangular pyramid
    angled_face = faces().sort_by(Axis.Y)[-1] # select some face
    with BuildSketch(angled_face) as s2: # create a 2D sketch on that face
        Circle(.2)
    extrude(s2.sketch,amount=.5) # extrude that sketch in a direction normal to the earlier face
```
[image]

The recommended workflow for build123d is a lot closer in concept to the other BREP modelers (SolidWorks, OnShape, FreeCAD, etc) than it is to OpenSCAD. This is in part because, unlike OpenSCAD, introspection of the existing model is possible as demonstrated in the last example. As a former avid user of OpenSCAD, it took me quite a long time to stop thinking along the lines of OpenSCAD and to start thinking along the lines of BREP modelers.

KilowattSynthesis, 2024-11-11:
I generally prefer the algebraic syntax, as the context manager syntax has a little bit too much magic, and it's not always clear what part you're operating on/transforming, and where the output will be stored when complete.

I agree that I have more learning to do though. Thanks for that demo/explanation!