gumyr/build123d #778: Adding EXTRABOLD to fon

startOfStars, 2024-11-11:
Hello

At the moment EXTRABOLD seems not be supported in fontStyle ?
I find REGULAR, BOLD, ITALIC but not EXTRABOLD.

Thank you !

gumyr, 2024-11-11:
`EXTRABOLD` is not supported by the OpenCascade CAD kernel that is used by build123d.  However, there are some unused values that could be added - these are the OCCT options:

- Font_FontAspect_UNDEFINED
- Font_FontAspect_Regular
- Font_FontAspect_Bold
- Font_FontAspect_Italic
- Font_FontAspect_BoldItalic
- Font_FA_Undefined
- Font_FA_Regular
- Font_FA_Bold
- Font_FA_Italic
- Font_FA_BoldItalic

while these are the options exposed by build123d:

- REGULAR
- BOLD
- ITALIC
 
At this point it isn't clear what the difference between the `FontAspect` and `FA` versions are but it seems like `BoldItalic` would be good to add.

startOfStars, 2024-11-11:
Thank you for your answer.
To be honest I don't understand all of what you wrote... 

I was asking this because for exemple for Noto Sans Google Font, 18 styles are available, including "ExtraBold".

PS : Thank you for this amazing build123d !

gumyr, 2024-11-11:
The fonts are provided by the CAD kernel not Google or anywhere else so we're restricted to what OCCT provides.

jwagenet, 2025-02-17:
@startOfStars A little late, but you can directly call the font with the `font_path` kwarg eg. `Text("Hello World", 10, font_path="C:/Windows/Fonts/NotoSans-ExtraBold.ttf")` on Windows. This seems to override both `font` and `font_style`. The docs don't really indicate this behavior.

jwagenet, 2025-04-11:
> At this point it isn't clear what the difference between the FontAspect and FA versions are

It seems like the FA versions are just aliases for the FontAspect versions

commit 9b78e076 (2025-05-01): FontStyle and and Text: add BOLDITALIC font aspect to resolve #778
closed 2025-05-01 by gumyr