gumyr/build123d #781: Regular polygon confusing alignment behavior

erooke, 2024-11-12:
The alignment behavior for `RegularPolygon` seems to be different than for other objects. From what I can see most objects get aligned by bounding box. However if you ask for `RegularPolygon` to be centered it gets aligned according to the circumscribed circle.

The left image is what happens if you give `align = Align.CENTER` while the right is what would normally happen.
[image]

Is this intended? The docs do not mention this but there are tests enforcing this behavior: https://github.com/gumyr/build123d/blob/47814adfa6b07ae85f69b98582560ef0c3ab94f8/tests/test_build_sketch.py#L296-L303

jdegenstein, 2024-11-13:
This is intentional behavior and we are unlikely to change it. This is documented on this page of the docs, including one example specifically for triangles. https://build123d.readthedocs.io/en/latest/center.html

As mentioned on that page, there is even an enum to control this: `CenterOf`.

erooke, 2024-11-13:
Fair enough, dunno how I missed that page in the docs.

closed 2024-11-13 by erooke