gumyr/build123d #785: add wire/edge methods `position_at, tangent_at, location_at` to `class Curve`

jdegenstein, 2024-11-13:
Currently the operators `@, %, ^` for `position_at, tangent_at, location_at` are bound to `class Curve` but the actual methods ``position_at, tangent_at, location_at`` are not present

gumyr, 2024-11-19:
Please hold off on this.  As part of the refactoring of topology.py (Issue #788) `Curve` is changing. Still needs some verification but a `Curve` that can be a `Wire` is converted to one during the `+` operation so all the normal `Mixin1D` methods work for them.  If the `Curve` consists of disconnected `Edges` then it will be a `Compound` and these operations don't apply.

jdegenstein, 2024-11-20:
Makes sense, and sounds good -- I started working on it briefly and then realized I was creating more entanglements so I stopped.

gumyr, 2025-05-15:
@jdegenstein I think we can close this issue now - do you agree?

closed 2025-05-15 by jdegenstein