gumyr/build123d #796: Positions with incorrect type should raise a TypeError

fouronnes, 2024-11-21:
As of 0.7.0 this code runs without error, and the sketch is not moved but stays at 0, 0:

```python
from build123d import *

sketch = Pos(0, "blip") * Circle(5)
sketch
```

We should add some type checking to raise a TypeError here. This can happens quite easily if the coordinate comes from another data type like numpy or such. It can pretty easily be wrapped into a np.array or something, then it's really confusing why no transform is applied.

gumyr, 2025-01-14:
The tests cover the following now:

```py
        with self.assertRaises(TypeError):
            Pos(0, "foo")
        self.assertEqual(Pos(1, 2, 3).position, Vector(1, 2, 3))
        self.assertEqual(Pos((1, 2, 3)).position, Vector(1, 2, 3))
        self.assertEqual(Pos(v=(1, 2, 3)).position, Vector(1, 2, 3))
        self.assertEqual(Pos(X=1, Y=2, Z=3).position, Vector(1, 2, 3))
        self.assertEqual(Pos(Vector(1, 2, 3)).position, Vector(1, 2, 3))
        self.assertEqual(Pos(1, Y=2, Z=3).position, Vector(1, 2, 3))
```

gumyr, 2025-01-14:
The interfaces on the base classes have been improved somewhat but there is no guarantee of generalized type checking.  If there are examples like this please raise another issue.

closed 2025-01-14 by gumyr