gumyr/build123d #801: build123d caching POC

vdp, 2024-11-25:
Hi,

as suggested in a [reddit discussion](https://www.reddit.com/r/build123d/comments/1gnvd4n/caching_build123d_objects/) this issue documents my brief investigation and proof-of-concept port of the cq_cache, which is CadQuery plugin that caches the build123d parts created by a Python function. To make this GH issue self-contained I'll include all the relevant code at the end of this message, and to save time I may copy/paste from the already mentioned Reddit thread.

The problem that made me think about caching is that adding bd_warehouse threads to a few holes on a part I was working on was taking more than 10 seconds to render, which quickly becomes annoying. I'm a build123d newbie, so it's entirely possible I was not using the threads in the most optimal manner. Anyway, I came across [cq_cache](https://github.com/CadQuery/cadquery-plugins/tree/main/plugins/cq_cache), and decided to see if I could modify it to work with build123d. Turns out it's really straightforward even though I'm not familiar with build123d's internals. The result, a simple test and timing results can be seen below. The implementation shown is just a quck-and-dirty mod of the original code, and there are likely lots of bugs, missing and redundant code.

Now, is something like this really needed? I'm not sure TBH. Because build123d benefits from the full strength and the rich ecosystem of Python, there are probably plenty of other ways to work around the geometry construction slowness if it becomes a problem. Some, I can think of at the moment, are:

* split the construction into different modules/functions, so that you can work on different facets of the design in isolation when possible

* make the "slow" geometry conditional on level-of-detail flags. For example, if adding threads to holes is slow, you can have a boolean such as ADD_THREADS and if it's set to False only the holes are shown without the threads

* in envirnoments such as jupyter notebooks, with partial execution/caching already built-in, additional caching layer would be redundant. I think in this case one can use Python's functools.cache as a Discord user commented about a week ago, or just assign the result to a variable in a preceding cell, and then shallow-copy it if there is need for more than one part instance.

I'm a CAD newbie and don't have enough 'data' to tell if there are situations where the proposed caching would be a clear winner. Apparently such situations, even if they exists, are quite rare. The fact that cq_cache itself is not updated for 3 or so years, even though it has plenty of scope for improvement, is telling.

My takeaway conclusion from this quick experiment is that such a persistent caching solution likely can't be entirely transparent to the user. That is, they can't just slap a '@build123d.cache' decorator before a part-making function and then forget all about it. The problem is that without some care the returned cache can become stale. In the case of build123d such caching solution would be likely most useful when iterating on a (complex/big/slow) design. Therefore I think it's crucial to have a robust way to detect changes in the code that must trigger cache invalidation. I wasn't completely successful in finding 100% foolproof method to detect changes in the cached function's code. A method based on inspect.getsource() looks promising and it worked when calling the script from the command line, but for some reason failed when cq-editor is used. Even if there is a good way to detect changes in decorated function's body, I believe it will not be 100% dependable unless the function is "pure" and doesn't depend on other(non-cached) functions and global variables. For example, my build123d scripts usually start with a parameters/constants section, and the subsequent code refers to them to parameterize the construction process. If a function whose result is cached refers to global variables to make a part and some of these parameters change, the currently considered simplistic approaches won't invalidate the cache, because the code of the function itself doesn't change. To be robust, the function must only use parameters that it receives as arguments. I can think of ways to semi-automate the passing of global state as function argument, but that won't be much less intrusive than using level-of-detail flags, for example, with the added disadvantage of being opaque to the user.

Sorry for the long write-up. As far as I understand the purpose of this GH issue is to serve as a reminder, so I wanted to make it as self-contained as possible, even at the risk of stating the obvious.

**bxd_cache.py**
```python
"""
Port of cq_cache for build123d
"""

import os
import sys
import tempfile
import inspect
import base64
import hashlib
from itertools import chain
from functools import wraps

from OCP.BRepTools import BRepTools
from OCP.BRep import BRep_Builder
from OCP.TopoDS import TopoDS_Shape

import build123d as bxd

TEMPDIR_PATH = tempfile.gettempdir()
CACHE_DIR_NAME = "build123d_geom_cache"
CACHE_DIR_PATH = os.path.join(TEMPDIR_PATH, CACHE_DIR_NAME)
BXD_TYPES = [
    bxd.Part,
    bxd.Shape,
    bxd.Solid,
    bxd.Shell,
    bxd.Compound,
    bxd.Face,
    bxd.Wire,
    bxd.Edge,
    bxd.Vertex,
    bxd.Plane,
    TopoDS_Shape,
]

if CACHE_DIR_NAME not in os.listdir(TEMPDIR_PATH):
    os.mkdir(CACHE_DIR_PATH)

#$ Doesn't quite work for our purposes b/c co_consts changes if there are, for
## example list comprehensions in the body of the cached function, even if the
## resulting list stays the same
## def hash_fn(fn):
##     """
##     Creates function signature, to be able to detect when its code is modified
##     """
##     fn_enc = str(fn.__code__.co_consts).encode('utf-8') + fn.__code__.co_code
##     return hashlib.md5(fn_enc).hexdigest()
[... 319 lines cut ...]
    part = thrbox_naive()
show_object(part.locate(Location((-35, 0, 0))), options={'alpha': .5})

with timeit_context("pre-threaded blocks construction"):
    part = thrbox_add_prethreaded()
show_object(part, options={'alpha': .5})

with timeit_context("partially cached construction"):
    part = thrbox_partial_cache()
show_object(part.locate(Location((35, 0, 0))), options={'alpha': .5})

with timeit_context("fully cached construction"):
    part = thrbox_fully_cached()
show_object(part.locate(Location((0, 35, 0))), options={'alpha': .5})
```

**timing results**(old laptop)
```
Elapsed time for "naive construction": 11.193s
Elapsed time for "pre-threaded blocks construction": 7.053s
loading "thrbox_block_cached_c421a9a6422544ed1158fde7a89fa6c2" from cache
Elapsed time for "partially cached construction": 4.794s
loading "thrbox_fully_cached_5d81c7dac0e07be6c1508d579703ad8c" from cache
Elapsed time for "fully cached construction": 0.028s
```

gumyr, 2024-11-25:
Thank you for your detailed look at object caching.  I've not looked into caching in detail but the Discord user "barnaby" posted the following:

```py
#%% Imports
from build123d import *
import gridfinity_build123d as gfb
from ocp_vscode import *
import functools

@functools.cache
def memoized_base(units_x, units_y):  # Arguments are used for dict keys and must be hashable!
  # gfb.Base is slow, taking about 6s to run for a 5x3 base!
  return gfb.Base(grid=((True,) * units_x,) * units_y,
    features=(gfb.MagnetHole()),
    align=(Align.MIN, Align.MIN, Align.MIN)
  )

#%% Part
with BuildPart() as part:
  # Slow on the first execution, instant on subsequent executions with the same arguments
  base = add(memoized_base(3, 4))
  # ...
  ```
However it's done, there should be a section in the docs describing these types of caching solutions.

@vdp note that bd_warehouse threads are designed as assemblies (`Compound`) to keep them small and fast.  In a bd_warehouse `Screw` only one of the central thread segments is created, the rest are shallow copies, which improves speed/size dramatically.  When using thread it's recommended that users also follow this approach and add thread to their parts by creating an assembly and not by fusing the thread into the part which is slow and often problematic at the CAD kernel level.

drwho495, 2025-04-01:
any updates on this?