gumyr/build123d #804: Feature Request: Add a `Part.scale_axes(vector)` function which allows scaling by different amounts on each axis

KilowattSynthesis, 2024-11-30:
It would be awesome if there was a way to just scale a single axis (e.g., squish the x-axis by 0.5, and keep the y-axis and z-axis at 1x scale).

Currently, it seems that the `Part.scale(float)` function scales all three axes at once.

An example of applying this function, where there exists no easy alternative, is making a cone that's squished on one axis.

gumyr, 2024-11-30:
This feature already exists:
```py
from build123d import *
from ocp_vscode import show_all

b = Box(1, 1, 1)
b.color = Color("Blue")
scaled = Pos(X=1.5) * scale(b, (1, 1, 0.5))

show_all()
```
[image]

In general users should be using the functions over the direct API methods as they have more features.

closed 2024-11-30 by gumyr

KilowattSynthesis, 2024-12-01:
> In general users should be using the functions over the direct API methods as they have more features.

I wish I could use the function, but it seems that only the class way is supported for this. 

This feature request is to add the feature that's lacking. Saying "use the other way because it has more features" is a weird way to respond to a report of a missing feature.

jdegenstein, 2024-12-05:
I believe what gumyr was referring to is that `scale` is a function and `Part.scale` is a method.
```py
a = Cone(1,0,2)
a = scale(a, (1,1,.5))
```
In general the objects and operations on the cheat sheet are what should be considered the user-facing APIs (builder and algebra). https://build123d.readthedocs.io/en/latest/cheat_sheet.html