gumyr/build123d #811: Bug in Compound location handling

MatthiasJ1, 2024-12-09:
```py
a = GridLocations(60, 30, 2, 2) * extrude(Rectangle(50, 20), -5)
b = pack([Compound(children=[a[0]]), *a[1:]], 1)
```

[image]

If you fuse b, it will be partially fixed afterwards which shows this is not a bug in `pack`.

gumyr, 2025-01-02:
Note for self - here is a hypothesis on what might be happening:

There were changes in the code in November that eliminated "needless" Compound wrappers around objects - a common occurrence with OCCT operations - where "needless" is defined as a `Compound` that has only one member. In this case the singleton Compound has a non zero Location so stripping that off changes where the enclosed shape is ultimately located.  

If this is the problem then this "unwrapping" procedure needs to re-locate the enclosed shape.

gumyr, 2025-01-02:
Maybe not:

```py
box = Compound(children=[Solid.make_box(1, 1, 1)]).locate(Pos((2, 0, 0)))
assembly = Compound(children=[box]).moved(Pos((0, 2)))
print(assembly.show_topology("Solid"))

unwrapped = unwrap_topods_compound(assembly.wrapped)
print(unwrapped, Location(unwrapped.Location()))
```
```
             Compound at 0x7b341b3a2a40, Location(p=(0.00, 2.00, 0.00), o=(-0.00, 0.00, -0.00))
└──          Compound at 0x7b341a5abdc0, Location(p=(2.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
    └──      Solid    at 0x7b341a5abe20, Location(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))

<OCP.TopoDS.TopoDS_Solid object at 0x7b341a5ded70> Location: (position=(2.00, 2.00, 0.00), orientation=(-0.00, 0.00, -0.00))
```
The resulting Solid is in the correct Location.

bernhard-42, 2025-01-29:
I can confirm that this is the viewer bug https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/145
Will publish a fix soon

bernhard-42, 2025-03-04:
@gumyr @MatthiasJ1 For the records, with the latest version of the viewer this now creates what is expected

[image]

closed 2025-05-14 by gumyr