gumyr/build123d #817: Add a DXF importer

gumyr, 2024-12-15:
Here is an initial version that needs to be finished: 
```py
"""
build123d import dxf

name: import_dxf.py
by:   Gumyr
date: November 10th, 2024

desc:
    This python module imports a DXF file as build123d objects.

license:

    Copyright 2024 Gumyr

    Licensed under the Apache License, Version 2.0 (the "License");
    you may not use this file except in compliance with the License.
    You may obtain a copy of the License at

        http://www.apache.org/licenses/LICENSE-2.0

    Unless required by applicable law or agreed to in writing, software
    distributed under the License is distributed on an "AS IS" BASIS,
    WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
    See the License for the specific language governing permissions and
    limitations under the License.

"""

import math
import warnings
import ezdxf
from build123d.objects_curve import (
    CenterArc,
    EllipticalCenterArc,
    Line,
    Polyline,
    SagittaArc,
    Spline,
    ThreePointArc,
)
from build123d.drafting import Arrow
from build123d.build_enums import Align
from build123d.objects_sketch import Circle, Polygon, Text
from build123d.geometry import Axis, Pos, TOLERANCE, Vector
from build123d.operations_generic import scale
from build123d.topology import ShapeList, Vertex, Wire

def process_arc(entity):
    """Convert ARC"""
    start, mid, end = entity.angles(3)
    arc_center = Vector(*entity.dxf.center)
    radius_vec = Vector(entity.dxf.radius, 0, 0)
    pnts = [arc_center + radius_vec.rotate(Axis.Z, a) for a in [start, mid, end]]
    return ThreePointArc(*pnts)

def process_circle(entity):
    """Convert CIRCLE"""
    return Pos(*entity.dxf.center) * Circle(entity.dxf.radius).edge()

def process_ellipse(entity):
    """Convert ELLIPSE"""
    center = entity.dxf.center
    major_axis = entity.dxf.major_axis
    x_radius = (major_axis[0] ** 2 + major_axis[1] ** 2) ** 0.5
    y_radius = x_radius * entity.dxf.ratio
    rotation = math.degrees(math.atan2(major_axis[1], major_axis[0]))
    start_angle = math.degrees(entity.dxf.start_param)
    end_angle = math.degrees(entity.dxf.end_param)
[... 198 lines cut ...]
    Returns:
        ShapeList: build123d objects
    """
    try:
        doc = ezdxf.readfile(filename)
    except ezdxf.DXFStructureError:
        raise ValueError(f"Failed to read {filename}")
    build123d_objects = []

    # Iterate over all entities in the model space
    for entity in doc.modelspace():
        dxftype = entity.dxftype()
        print(f"{dxftype=}")
        if dxftype in entity_dispatch:
            new_object = entity_dispatch[dxftype](entity)
            print(f"{new_object=}")
            if isinstance(new_object, list):
                build123d_objects.extend(new_object)
            else:
                build123d_objects.append(new_object)
        else:
            warnings.warn(f"Unable to convert {dxftype}")

    return ShapeList(build123d_objects)
```

petejohanson, 2024-12-22:
Looking forward to this. Just added this locally to a new project I'm working on, and this properly imported the DXF files exported by ergogen just fine for some case generation work.

gumyr, 2024-12-23:
Thanks for trying it out - good to know it works at least somewhat.  IIRC DXF polyline may have a "bump" which needs to be added along with tests etc.