gumyr/build123d #818: running example fails on windows (RuntimeError: Unable to open config file C:\Users\user\.ocpvscode)

edrihan, 2024-12-16:
I did `pip install build123d`
Then I copy pasted the canada flag example into a file and ran it. This is the output:

`Lock file C:\Users\user\.ocpvscode.lock already exists
Lock file C:\Users\user\.ocpvscode.lock is stale
Traceback (most recent call last):
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\state.py", line 156, in get_state
    with open(config_file, "r", encoding="utf-8") as fd:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\.ocpvscode'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\user\Desktop\3dcanadaflag.py", line 105, in <module>
    show(Rot(90, 0, 0) * canadian_flag)
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\show.py", line 555, in show
    t, mapping = _convert(
                 ^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\show.py", line 255, in _convert
    instances, shapes, config, count_shapes, mapping = _tessellate(
                                                       ^^^^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\show.py", line 86, in _tessellate
    if workspace_config().get("_splash"):
       ^^^^^^^^^^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\config.py", line 431, in workspace_config
    port = get_port()
           ^^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\comms.py", line 111, in get_port
    find_and_set_port()
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\comms.py", line 302, in find_and_set_port
    port = find_port()
           ^^^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\comms.py", line 266, in find_port
    states = get_state().items()
             ^^^^^^^^^^^
  File "C:\bin\Python\Python311\Lib\site-packages\ocp_vscode\state.py", line 164, in get_state
    raise RuntimeError(f"Unable to open config file {config_file}.") from ex
RuntimeError: Unable to open config file C:\Users\user\.ocpvscode.`

gumyr, 2024-12-17:
`ocp_vscode` is a separate project from `build123d` - you could raise an issue here: https://github.com/bernhard-42/vscode-ocp-cad-viewer or as for assistance in the CadQuery/build123d [Discord](https://discord.com/invite/Bj9AQPsCfx).

A couple things to try though:
1) Open a Folder in vscode (File->Open Folder)
2) Close vscode entirely and re-open it

jdegenstein, 2024-12-17:
Assuming you are running ocp_vscode in standalone mode, this is a known issue and documented on the appropriate repo https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/121