gumyr/build123d #831: Creating Gordon Surfaces

gumyr, 2024-12-20:
### Discussed in https://github.com/gumyr/build123d/discussions/830

<div type='discussions-op-text'>

<sup>Originally posted by **jsbroks** December 19, 2024</sup>
I'm wondering what is the best way to create a Gordon Surface?

Here is what I have so far but as you would expect I getting `Error building a non-planer face with the provided exterior`.

```python
import csv
import numpy as np
from build123d import *

def read_heightmap(filename: str):
    with open(filename, 'r') as f:
        reader = csv.reader(f)
        heightmap = []
        for row in reader:
            heightmap.append([float(val) for val in row])
    return np.array(heightmap)

heightmap = read_heightmap('heightmap.csv')

points = []
for x in range(heightmap.shape[0]):
    for y in range(heightmap.shape[1]):
        points.append(Vector(x, y, heightmap[x,y]))

x_splines = []
y_splines = []

for x in range(heightmap.shape[0]):
    points = [(x, y, heightmap[x,y]) for y in range(heightmap.shape[1])]
    x_splines.append(Spline(points))

for y in range(heightmap.shape[1]):
    points = [(x, y, heightmap[x,y]) for x in range(heightmap.shape[0])]
    y_splines.append(Spline(points))

boundary_wire = Wire.make_wire([x_splines[0], y_splines[0], x_splines[-1], y_splines[-1]])

show_object(Face.make_surface(boundary_wire))
```</div>

gumyr, 2024-12-20:
From @mnesarco

TiGL[1] has an OpenCascade based Gordon surface builder that is very good (c++). Christophe Grellier (@tomate44) ported it to FreeCAD's Curves WB[2] long time ago in Python. It can be used as a reference to port it to build123d. I am in the process of reorganizing the implementation as a standalone python module, but it is WIP and i can't make compromises on that [3]

[1] https://dlr-sc.github.io/tigl/
[2] https://github.com/tomate44/CurvesWB/blob/master/freecad/Curves/gordon.py
[3] https://github.com/mnesarco/MarzWorkbench/blob/master/freecad/marz/curves/gordon.py

gongfan99, 2025-10-02:
@gumyr I have created a pull request to add the gordon surface. I have followed everything in the CONTRIBUTING.md

The only error is a type error:
src\build123d\topology\two_d.py:887: error: Argument 1 to "Curve_s" of "BRep_Tool" has incompatible type "Any | None"; expected "TopoDS_Edge"  [arg-type]
src\build123d\topology\two_d.py:888: error: Argument 1 to "Curve_s" of "BRep_Tool" has incompatible type "Any | None"; expected "TopoDS_Edge"  [arg-type]

It appears that this is a common error because Edge.wrapped has the "Unknown | None" type.