gumyr/build123d #843: `location_at` returns forward direction even when edge is reversed

jdegenstein, 2025-01-03:
credit to @snoyer for recommending this

Consider changing `location_at` so that it returns the actual positive direction even when the edge/wire is reversed. Have not looked at faces to see if they suffer similar a problem but worth investigating that further.

snoyer, 2025-01-03:
Supporting visual elements from the initial investigation on Discord with @jdegenstein

1. Visualizing edge direction using locations:
[image]

2. However, looking at `.is_forward`:
[image]
```py
cone = Cone(0.3, 0, 0.8)
part = Box(10, 10, 2) - Cylinder(3, 2)
show_object(part.wrapped)
for e in part.edges():
    show_object(
        (e ^ 0.5) * cone,
        options=dict(color=(1.0, 0.0, 0.0) if e.is_forward else (0.0, 0.0, 1.0)),
    )
```

3. now with locations accounting for `.is_forward`:
[image]

CarpeNecopinum, 2025-01-06:
My current use case for this is optimizing tool paths, which was surprisingly slow due to how `Edge.reversed()` works (5s for 900 flips, vs. 290µs with `Edge.wrapped.Reverse()`)

We should look into whether we can replace the implementation of `Edge.reversed()` once features like `.position_at` and `start_point`/`end_point` respect the `is_forward` flag.

closed 2026-10-14 by gumyr