gumyr/build123d #850: Enhancement of the make_brake_formed() function, and support of stamping/forming of sheetmetal and automatic creation of dies for stamping sheetmetal

kcleung, 2025-01-07:
Have a look at the FreeCAD sheetmetal workbench:

https://wiki.freecad.org/SheetMetal_Workbench

It would be great if the build123d make_brake_formed() can also have advanced folding functions such as making folds with a specified bend radius, relief, and corner relief

Furthermore, it would be great if we also have a function of stamping of sheetmetal, and automatic creation of a die set for stamping of sheetmetal.  This would be very useful for applications such as compound curves and combined cut and form (e.g. louvre)

kcleung, 2025-01-25:
Upon reflection, the designing of the stamping mechanism, including the die, also needs to consider physical property of the material, in particular to allow the sprining of the material.

Therefore, this feature will also depend on the material system being implemented in build123d #598

jdegenstein, 2026-10-07:
Update: BuildSheet has been added which greatly enhances sheet metal capability in build123d. See docs here https://build123d.readthedocs.io/en/latest/sheet_metal/index.html

Also, a material system was added as well. See docs here https://build123d.readthedocs.io/en/latest/tutorial_materials.html#material-tutorial

Regarding this proposal (automatic creation of dies for stamping sheetmetal), this seems to fall much more into the manufacturing side which I am not sure has a lot of precedent in build123d. For example consider the CAD/CAM separation in most other CAD systems. Regarding the FreeCAD Sheet Metal Workbench -- it also does not have this capability.

I believe this capability is currently only present in commercial products like Siemens NX Progressive Die Design or Logopress for SolidWorks.