gumyr/build123d #851: Automatic creation of moulds for casting

kcleung, 2025-01-07:
Given a 3D solid model, it would also be nice to have a function for automatic creation of a set of moulds and cores for casting, with gates etc.

This function should automatically work out the cores required, gates required, take into consideration of shrinkage ratio supplied by the user.

Jojain, 2025-01-08:
While things like that seems great, the amount of work required to do so would fit its own project. I'm not sure `build123d` can handle that. It's mainly a modelling lib, not a full blown CAD app.

kcleung, 2025-01-24:
> This seems like a very ambitious project but I'll leave it here as someone might be able to take it on in the future.

@gumyr @cadquery Have a look at the third screenshot from the cq_editor page:

https://github.com/CadQuery/CQ-editor

It shows a set of dies to be cast, though I don't know whether the user made the die manually, or there exists a cadquery programme that takes a cadquery part together with material details, and use these details to allow for shrinkage, and insert gates.

Also have a look at cinmatron, which provide automated design of stamping/folding dies and casting molds:

https://www.cimatron.com/en 

Anyone has experience in its workflow?

gumyr, 2025-01-25:
I believe the die showed there was made manually, CadQuery doesn't have the ability to automatically do this.

@kcleung are you offering to create this feature?

kcleung, 2025-01-25:
> I believe the die showed there was made manually, CadQuery doesn't have the ability to automatically do this.
> 
> [@kcleung](https://github.com/kcleung) are you offering to create this feature?

Would love to, but I first need to be fluent at build123d, as well as be very familiar with workflow of existing automated die and mold design software first so that we can properly extend the automated die and mould design workflow to build123d