gumyr/build123d #864: Wire.edges() should use BRepTools_WireExplorer

gumyr, 2025-01-13:
@sno141421
TopoDS_Explorer iterates all the edges according to the internal storage order (whatever that is ) while BRepTools_WireExplorer iterates the continuous edges according to the curve order (also whatever that is  )

snoyer, 2025-01-13:
to complete the quote:

> not sure if you need, or even want, `BRepTools_WireExplorer` in the general `.edges()` case (as it may miss some edges in degenerate wires) but for the SVG export it looks like _not_ using it _may_ (confirmation needed) be causing issues

gumyr, 2025-02-20:
Updated as follows:
```py
    def edges(self) -> ShapeList[Edge]:
        """edges - all the edges in this Shape"""
        if isinstance(self, Wire):
            # The WireExplorer is a tool to explore the edges of a wire in a connection order.
            explorer = BRepTools_WireExplorer(self.wrapped)

            edge_list: ShapeList[Edge] = ShapeList()
            while explorer.More():
                edge_list.append(Edge(explorer.Current()))
                explorer.Next()
            return edge_list
        else:
            edge_list = Shape.get_shape_list(self, "Edge")
            return edge_list.filter_by(
                lambda e: BRep_Tool.Degenerated_s(e.wrapped), reverse=True
            )
```

closed 2025-02-20 by gumyr