gumyr/build123d #866: `pip install build123d` is broken for version `0.8.0`

jdegenstein, 2025-01-13:
Currently 
```
pip install build123d
```
Will install `build123d==0.8.0` and `cadquery-ocp==7.8.1.0` -- the problem is that OCCT/OCP 7.8 introduced incompatibilities in the API that affect build123d which did not have strict enough version requirements for 0.8.0 and below. This means that build123d will not work correctly.

The workaround:
```
pip install build123d "cadquery-ocp<7.8"
```
This problem will be corrected soon in a future release of build123d.

gumyr, 2025-01-14:
I'm not sure if this is gating or not.  Are we going to support older versions of OCP for an extended length of time?

jdegenstein, 2025-01-14:
Sorry my point was to say that this needs to be fixed soon. Maybe that tag was not appropriate for this.

42sol-eu, 2025-01-20:
I can test updazes on MacOS for intel and arm64 if it helps

contagnas, 2025-01-26:
I had trouble finding this issue in search so adding an error this caused for me:

```
    out[item.HashCode(HASH_CODE_MAX)] = (
        ^^^^^^^^^^^^^
AttributeError: 'OCP.OCP.TopoDS.TopoDS_Shape' object has no attribute 'HashCode'
```

jdegenstein, 2025-01-26:
Thank you @contagnas and sorry for the oversight!

Rapter35, 2025-01-29:
Thank you @contagnas for making it searchable, I encountered the same issue after installing build123d on a new computer.
```
out[item.HashCode(HASH_CODE_MAX)] = (
        ^^^^^^^^^^^^^
AttributeError: 'OCP.OCP.TopoDS.TopoDS_Shape' object has no attribute 'HashCode'
```

@jdegenstein the workaround you proposed of `pip install build123d "cadquery-ocp<7.8"` returns this error for me:

```
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.8 (from versions: 7.8.1.0, 7.8.1.1, 7.8.1.1.post1)
ERROR: No matching distribution found for cadquery-ocp<7.8
```
Pip is version `pip-25.0`, in case that helps.

Thank you.

CePeU, 2025-02-01:
I just installed it with my portable version and it worked/works. For the install pip-24.xx was used. I upgraded to pip 25.0 after the cadquery update.

System is Windows with a seperate portable VSCode and python setup.

[image]

jdegenstein, 2025-02-02:
@Rapter35 build123d==0.8.0 and lower do NOT support python 3.13 (they only support python 3.12). You can either use older python or newer build123d from github.

jdegenstein, 2025-02-05:
FYI, build123d 0.9.0 was released yesterday to PyPI. I plan to close this issue in the next few days as we are not planning to fix the older releases of build123d.

https://github.com/gumyr/build123d/releases/tag/v0.9.0

closed 2025-02-06 by jdegenstein