gumyr/build123d #871: Upgrading from cadquery-ocp<7.8 to cadquery-ocp>=7.8

jdegenstein, 2025-01-15:
Reproduced a user report that during `from build123d import *` an error occurred: `ImportError: cannot import name 'IVtkOCC_Shape' from 'OCP.IVtkOCC' (...\Lib\site-packages\OCP\IVtkOCC\__init__.py)`

Traced it down and determined that if an environment previously contained e.g. `cadquery-ocp==7.7.2.1` then upgrading to e.g. `cadquery-ocp==7.8.x` will cause this problem. I suspect that some of the VTK libraries in the older version are not uninstalled completely and are creating some conflicts.

The workaround is create a new clean environment from scratch and directly install the newer version of build123d (and its dependency `cadquery-ocp>7.8`).

jdegenstein, 2025-01-16:
Another workaround if the installation is broken:
```
pip uninstall vtk
pip install vtk==9.3.1
```

gumyr, 2025-05-15:
@jdegenstein can we close this one?  I haven't seen this come up in a while.

jdegenstein, 2025-05-15:
I prefer to keep this one open for a bit longer as there are still a number of projects out there pinned to v0.8.0 for some reason.