gumyr/build123d #881: "3mf mesh is invalid" from imported SVG

adamcrume, 2025-01-27:
Simply extruding one of the shapes from https://rustacean.net/assets/rustacean-flat-noshadow.svg and meshing it raises an error.

```python
from build123d import Mesher, extrude, import_svg

shapes = import_svg("rustacean-flat-noshadow.svg")
Mesher().add_shape(extrude(shapes[3], 1))
```

```
Traceback (most recent call last):
  File "/home/adam/Documents/3d-printing/badge flair build123d/src/bug.py", line 16, in <module>
    Mesher().add_shape(extrude(shapes[3], 1))
  File "/home/adam/apps/build123d/lib/python3.12/site-packages/build123d/mesher.py", line 442, in add_shape
    raise RuntimeError("3mf mesh is invalid")
RuntimeError: 3mf mesh is invalid
```

snoyer, 2025-01-27:
`shapes[3]` comes from a self-intersecting path that would be a valid Wire but does not produce a valid Face
```py
[w.is_valid() for w in shapes[3].wires()] = [True]
shapes[3].is_valid() = False
```

https://github.com/user-attachments/assets/07a69fec-1784-4132-a181-74a0a516807c

jdegenstein, 2025-01-27:
Related issue https://github.com/gumyr/build123d/issues/842 with some discussion and advice

adamcrume, 2025-01-28:
OH MY GOSH, no wonder I didn't see that, it's so tiny.  Thanks!  Is there a way to split apart the wire in code so that it would generate valid faces (i.e. if I want this to work for other SVGs)?  #842 doesn't have much detail.

gumyr, 2025-01-29:
Yes, both `Edge` or `Wire` have `trim` methods that can be used to shorten them but finding the right `Edge`/`Wire` and the right place and rebuilding the shape with the shorten shape look to be challenging.