gumyr/build123d #884: `Location(translation=(10,0,0), rotation=(0,30,0))` -> unexpected keyword argument 'translation'

jdegenstein, 2025-01-28:
`Location(translation=(10,0,0), rotation=(0,30,0))` -> `Location.__init__()` got an unexpected keyword argument 'translation'

This is despite following the overload function signature provided in VSCode.

GarryBGoode, 2025-01-28:
I had a look at `Location` source. True, it only takes `*args` and no `**kwargs`. There are similar patterns that use both arg and kwarg, like the `_parse_intersect_args` function, but I don't see a way to implement that for `Location`, it feels super complicated to work through.

gumyr, 2025-01-28:
The orientation part of a `Location` can be set as another tuple as shown here:
```py
l1 = Location((1, 1, 0), (0, 0, 0))
l2 = Location((1, -1, 0), (0, 0, 90))
```

[image]

Note how `l1` is orientated the same way as the normal coordinate system. `l2` has a rotation about Z of 90 degrees so now the red line (previously the X location axis) is now in the direction of the global Y axis.

Input parsing of Location needs to be reworked as described in Issue #441.

gumyr, 2025-05-17:
Location takes kwargs as of 67115111e2139540fff5560fceaf7ade9a1c3ac8

closed 2025-05-17 by gumyr