gumyr/build123d #885: RigidJoint inconsistent results

MonadicT, 2025-01-28:
I am using RigidJoint to build a structure and have encountered an issue. Here is some sample code to illustrate it.

```
def mkbox(w,d,h):
    c = Box(w,d,h)
    RigidJoint("topright", c, Location((c.length/2, 0, h/2), 0))
    RigidJoint("botright", c, Location((c.length/2, 0, -h/2), 0))
    RigidJoint("topleft", c, Location((-c.length/2, 0, h/2), 0))
    RigidJoint("botleft", c, Location((-c.length/2, 0, -h/2), 0))

    return c

ts = mkbox(36, 3, 2)
bs = mkbox(36, 2, 3)
ll = mkbox(4, 4, 34)
rl = mkbox(4, 4, 34)

connect = "top"
# connect = "bot"
# connect = "both"

if connect == "top" or connect == "both":
    ts.joints["topleft"].connect_to(ll.joints["topright"])
    ts.joints["topright"].connect_to(rl.joints["topleft"])
if connect == "bot" or connect == "both":
    bs.joints["botleft"].connect_to(ll.joints["botright"])
    bs.joints["botright"].connect_to(rl.joints["botleft"])

show_object([ts, bs, ll, rl])
```

The result is what I expect when I connect the left leg, top stretcher, and right leg (the first attached figure). The result is as expected when I connect the left leg, bottom stretcher, and right leg. However, when I connect the top and bottom stretchers, the positioning of the top stretcher is wrong. I also don't understand why the two stretchers  are always together.
 
Am I doing something wrong here or is it a bug in build123d? Thanks for any explanation.

[image]
[image][image]

gumyr, 2025-01-28:
The objects that are moved by `connect_to` aren't fixed in that location so you need start with a fixed object and go around connected all the others to each other like this:
```py
ll.joints["topright"].connect_to(ts.joints["topleft"])
ts.joints["topright"].connect_to(rl.joints["topleft"])
rl.joints["botleft"].connect_to(bs.joints["botright"])

show(ll, ts, rl, bs, render_joints=True)
```
If any object appear twice as a parameter to `connect_to` its first position will be replaced with the last position.
[image]

Does this answer your question?

MonadicT, 2025-01-28:
I understand now. Thank you very much.

closed 2025-01-28 by gumyr